OpenAADL/ocarina #25: POLYORB-HI-C

laukokhao, 2015-01-30:
Hi

When using Ocarina to generate code from an AADL model, is it necessary to use the PolyORB-HI backend? I have managed to parsed and instantiated my AADL model. However, I could only generate code WITH the PolyORB backend, for e.g.:

/usr/local/bin/ocarina -f -aadlv2 -g polyorb_hi_c -r motor_system.impl pololu_motor.aadl

Are we able to just generate and build the code using -b?

In addition, if PolyORB is really necessary to generate code from an AADL model, what are the proepr ways to remove the po-hi header/object files thereafter?

--
Peter

yoogx, 2015-01-30:
Ocarina has three backends for code generation, PolyORB-HI/Ada, PolyORB-HI/C and POK. These runtimes provides necessary support to implement the semantics of AADL. Supported constructs varies from one backend to another, so one model can work with one, but not the other.

-b is for compiling it, -z to clean code

closed 2015-01-30 by yoogx

laukokhao, 2015-02-04:
Hi

After I had installed the "polyorb-hi-c-master" from https://github.com/yoogx/polyorb-hi-c , I attempted to run ./configure, however this error pops out:

devmt@AGI-OptiPlex-755:~/Downloads/polyorb-hi-c-master$ ./configure
configure: error: cannot find install-sh, install.sh, or shtool in "." "./.." "./../.."

Upon googling, I found this result http://askubuntu.com/questions/27677/cannot-find-install-sh-install-sh-or-shtool-in-ac-aux . I am able to run "libtoolize --force" successfully. However upon running "aclocal", this error pops out:

devmt@AGI-OptiPlex-755:~/Downloads/polyorb-hi-c-master$ libtoolize --force
libtoolize: putting auxiliary files in AC_CONFIG_AUX_DIR, `support'.
libtoolize: linking file `support/ltmain.sh'
libtoolize: putting macros in AC_CONFIG_MACRO_DIR, `m4'.
libtoolize: linking file `m4/libtool.m4'
libtoolize: linking file `m4/ltoptions.m4'
libtoolize: linking file `m4/ltsugar.m4'
libtoolize: linking file `m4/ltversion.m4'
libtoolize: linking file `m4/lt~obsolete.m4'
devmt@AGI-OptiPlex-755:~/Downloads/polyorb-hi-c-master$ aclocal
configure.ac:82: warning: macro 'AM_PATH_OCARINA' not found in library

Googling does not yield any results/solutions for this error, and I have been troubleshooting for the past 2 weeks.

Just to give you an idea of my system, upon running "echo $PATH":

devmt@AGI-OptiPlex-755:~/Downloads/polyorb-hi-c-master$ echo $PATH
/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games:/home/devmt/bin:/bin

--Peter

yoogx, 2015-02-04:
First, run ./support/reconfig to bring in missing files

Then, ensure Ocarina is in your PATH. If the script does not find it, it might be a good indication your path  is incomplete.