OpenAADL/ocarina #27: can not configure

SancSib, 2015-02-19:
Hi, 
I can not install the program. ./configuration does not work. I had already installed gnatgcc. In bin folder, there is just ocarina-config but I dont know how I can create configuration and make files. 

I just started to use ocarina. Thank you.

yoogx, 2015-02-20:
./configuration does not exist, ./configure does. Please read this file for more details
https://github.com/OpenAADL/ocarina/blob/master/INSTALL

Binary snapshots of Ocarina are available on http://www.openaadl.org

closed 2015-02-20 by yoogx

SancSib, 2015-02-20:
I had already read the install file. But when write ./configure , it turns
me error : there is no file configure . Also there is no make file in bin
folder.

yoogx, 2015-02-20:
As explained in the INSTALL file

"If you modify source files, build Ocarina after a checkout or make
distclean, or the directory hierarchy of the source files, you should
re-generate autoconf and automake files (configure, Makefile.in...);
to do this, from the main directory, run:

./support/reconfig"

So, to generate configure, run ./support/reconfig ..

SancSib, 2015-02-20:
bash: ./support/reconfig: No such file or directory

I will take such an error.

yoogx, 2015-02-20:
How did you checkout Ocarina? and from which directory are you executing this command?

SancSib, 2015-02-20:
In the ocarina folder and also in bin folder. But it does not configure in
both of them.

sibel@sibel-VGN-NS21Z-S:~/Applications
*/ocarina-2.0w-suite-i686-pc-linux-gnu-2014121*8$ ./configure
bash: ./configure: No such file or directory
sibel@sibel-VGN-NS21Z-S:~/Applications
*/ocarina-2.0w-suite-i686-pc-linux-gnu-20141218/bin*$ ./configure
bash: ./configure: No such file or directory
sibel@sibel-VGN-NS21Z-S:~/Applications/
*ocarina-2.0w-suite-i686-pc-linux-gnu-20141218/bin*$ ./support/reconfig
bash: ./support/reconfig: No such file or directory

I also copy the bin file into the directory : /user/local/bin

sibel@sibel-VGN-NS21Z-S*:/usr/local/bin*$ ./configure
bash: ./configure: No such file or directory
sibel@sibel-VGN-NS21Z-S*:/usr/local/bin*$ ./support/reconfig
bash: ./support/reconfig: No such file or directory

On the other hand, when I check if the program installed or not and write
"ocarina --version "
 from home directory, I can get the version number of the ocarina.

yoogx, 2015-02-20:
So you have the binary release of Ocarina, for which there is nothing do compile, simply use Ocarina as detailed in its User's Guide, visible here http://www.openaadl.org/doc/ocarina.html/

SancSib, 2015-02-20:
If I understand your reply correctly, this means that the installation has
already been done. so there is no need to make configure. I think that I
can start to use ocarina for our models.

Thank you for your reply.

Best

yoogx, 2015-02-20:
Yes, you installed the binary release, meaning that all you have to do is run ocarina from the command line, or use the Eclipse integration plug-in

SancSib, 2015-02-20:
Ok. thank you so much for your reply.

Best