OpenAADL/ocarina #31: Schedule_Window_Record_Term_Array definition

juli1, 2015-04-20:
The Get_Module_Schedule_Property returns a Schedule_Window_Record_Term_Array
which has the following definition

   type Schedule_Window_Record_Term is record
      Partition : Name_Id;
      Duration : Time_Type;
      Periodic_Processing_Start : Boolean;
   end record;

However, the partition part should be a node id with a reference to the partition instead of a name_id. This is a semantic error and it is not possible to solely rely on the name: looking for a subcomponent with this name in the component hierarchy is confusing because several sub-components can have this name.

Thus, it woul dbe better to put a Node_Id and reference the existing referenced component.

juli1, 2015-04-30:
ping!

yoogx, 2015-05-16:
I told you it would take time, as this ticket is dependent on issue #17, which is quite complex. An initial support is now available, and I updated the does and vxworks653 accordingly, ADIRU example passes fine. I close the ticket

closed 2015-05-16 by yoogx