OpenAADL/ocarina #46: constraint error raised

juli1, 2015-07-24:
When trying to generate code, I got the following error:

ocarina -aadlv2 -f -b -r producer_consumer::pc.flowimpl -g polyorb_hi_c producer-consumer.aadl
========================== OCARINA BUG DETECTED =========================+
| Detected exception: CONSTRAINT_ERROR                                    |
| Error: g-table.adb:223 overflow check failed                            |
| Please refer to the User's Guide for more details.                      |
+=========================================================================+

Exception name: CONSTRAINT_ERROR
Message: g-table.adb:223 overflow check failed
Call stack traceback locations:
0x49725a 0x496f11 0x6e25e4 0x6e18cb 0x6112fc 0x60fada 0x60fa40 0x60fa73 0x60fa34 0x4962e3 0x493676 0x4d6299 0x406750 0x405acd 0x7f63b2fa7b43 0x4033bb 0xfffffffffffffffe

This is the model I am using is on https://gist.github.com/juli1/cc2bdef8ae9dd538084b

yoogx, 2015-07-24:
As far as I can see, this is a *huge* model. It seems you hit a limit I've never seen before in GNAT.Table that we use to store elements. 

I need to investigate this in more depth. Which GNAT version did you use to compile Ocarina, this will help me knowing which line 223 is actually in your setup

juli1, 2015-07-27:
For now, GNAT GPL 2014. This model is used to evaluate performance of generated code, so yes, this is not a realistic model. On the other hand, a user-friendly message could be reported (e.g. "Ocarina does not handle model with more than XXX entities - with XXX being a gross evaluation of what GNAT can handle).

yoogx, 2017-01-07:
No need to address this request for the moment. Closing

closed 2017-01-07 by yoogx