OpenAADL/ocarina #48: ocarina fails to build with gcc-4.8

adegroote, 2015-08-03:
It fails with the following error:

```
gnatmake \
          -j4 -P ocarina-config.gpr -XBUILD=release -XLIBRARY_TYPE=static \
          -XABS_SRC_DIR="/home/adegroot/ocarina/src" \
          -XABS_BUILD_DIR="/home/adegroot/ocarina/src" \
          -aP../projects -aP../projects;
gcc-4.8 -c -gnatec=/home/adegroot/ocarina/src/../projects/restrictions.adc -O2 -fPIC -gnat12 -gnata -gnaty -gnatyu -gnatys -gnatpn -I- -gnatA /home/adegroot/ocarina/src/config/ocarina-configuration.adb
ocarina-configuration.adb:257:30: "Compilation_Date" not declared in "Source_Info"
ocarina-configuration.adb:258:27: "Compilation_Time" not declared in "Source_Info"
gnatmake: "/home/adegroot/ocarina/src/config/ocarina-configuration.adb" compilation error
```

From my basic ada understanding,  GNAT.Source_Info from FSF only provides File, Line, Source_Location and Enclosing_Entity).

yoogx, 2015-08-03:
You need either gcc 5.1 or GNAT GPL 2015 (from libre.adacore.com), this function has been added recently. Note you would also need this update due to forthcoming patches.

closed 2015-08-03 by yoogx

adegroote, 2015-08-03:
Thanks. Can you update doc/installation.rst to reflect that ?