OpenAADL/ocarina #53: OCARINA parse by default AADLv1 files

pgaufillet, 2015-11-12:
Nowadays, the active AADL version is 2.x and OCARINA should be aligned on it. It may be solved by a build configuration switch looking like:
```shell
./configure --aadlv[12] --aadlv[12]-by-default
```
or more simple, keep only the second argument:
```shell
./configure --aadlv2-by-default
```
and embedd systematically aadl v1 and aadl v2 parsers.

yoogx, 2015-11-12:
Ocarina embeds both parser all the time
ocarina -aadlv2 -> use v2 parser
ocarina -aadlv1 -> use v1 parser
./configure --aadlv2 -> v2 parser default, no need to add -aadlv2
./configure defaults to ocarina -aadlv1

pgaufillet, 2015-11-13:
Thanks for these precisions. That's what I thought I had done (and ```./config.status --config``` confirmed it) but it seems something had gone wrong somewhere. ```make distclean; ./support/reconfig; ./configure --...; make install``` solved the issue.

Autoconf should propose anyway AADLv2 by default as AADLv1 is almost not used anymore.