OpenAADL/ocarina #61: Support of nested system hierarchy

yoogx, 2016-02-13:
PolyORB-HI/C and PolyORB-HI/Ada backends do not correctly handle situations where systems are inside systems, so as to model deep hierarchy.

yoogx, 2016-02-13:
The patch above solve the issue for PolyORB-HI/C

nvcyc, 2021-01-14:
The `aadl_xml` backend has the same issue that it does not support nested system hierarchy.
May I know if you have a plan for fixing it in `aadl_xml`?
Let me know if you'd like me to open a new issue for the `aadl_xml` backend.

yoogx, 2021-01-14:
Can you provide a test for the failure you observe?

nvcyc, 2021-01-14:
Sure.

Assuming the test AADL file is named `Test_Nested_System.aadl` shown as follows:
```
package Test_Nested_System
public
    system deployment
    end deployment;

    system implementation deployment.impl
        subcomponents
            sec_sys: system second_layer_system.impl;
    end deployment.impl;

    system second_layer_system
    end second_layer_system;

    system implementation second_layer_system.impl
    end second_layer_system.impl;
end Test_Nested_System;
```

Then, with using the following command (with the `aadl_xml` backend):
`ocarina -aadlv2 -g aadl_xml -r Test_Nested_System::deployment.impl Test_Nested_System.aadl`
we will see the following error message:
```
+========================== OCARINA BUG DETECTED =========================+
| Detected exception: SYSTEM.ASSERTIONS.ASSERT_FAILURE                    |
| Error: ocarina-backends-xml_tree-nodes.adb:399                          |
| Please refer to the User's Guide for more details.                      |
+=========================================================================+

raised SYSTEM.ASSERTIONS.ASSERT_FAILURE : ocarina-backends-xml_tree-nodes.adb:399

Symbolic Traceback:
```
Note that the compilation passes without an error if the system component `second_layer_system` is declared as abstract component (or any other types of components.)

yoogx, 2021-01-14:
Bug solved, see commit in this issue

nvcyc, 2021-01-14:
Thanks! 
Just verified that the latest code with the fix works as expected.

nvcyc, 2021-01-14:
For this fix in `aadl_xml`, I'm wondering if the following line should also have the same, additional condition check:
https://github.com/OpenAADL/ocarina/blob/1b28ef0a3a05721b404665beac7e0800e75927c2/src/backends/aadl_xml/ocarina-backends-aadl_xml-main.adb#L556
and have `Root_System_Processed` toggled in this block?

In fact, would it be simpler to just have the following condition check:
```
if Category = CC_System and E = Root_System_Node then
```
in both [#176](https://github.com/OpenAADL/ocarina/blob/1b28ef0a3a05721b404665beac7e0800e75927c2/src/backends/aadl_xml/ocarina-backends-aadl_xml-main.adb#L176) and [#L556](https://github.com/OpenAADL/ocarina/blob/1b28ef0a3a05721b404665beac7e0800e75927c2/src/backends/aadl_xml/ocarina-backends-aadl_xml-main.adb#L556) without an introduction of the new variable `Root_System_Processed`?

yoogx, 2021-01-14:
Yes, you are right, that would simplify the codebase