OpenAADL/ocarina #69: dependecy on libwinpthread-1.dll

juli1, 2016-07-28:
Hello,

When trying to use the latest windows build, I got the following issue (see screenshot). Is there any additional required dependency to install to get it working?

Thanks.
[image]

yoogx, 2016-07-28:
Probably a misconfiguration on my end, I may have compiled for cygwin, and so you miss this dependency. Furthermore, this binary is ad hoc and quite old. I'll see if I can generate a new one.

yoogx, 2016-08-02:
Indeed, I got issues using the "wrong" compiler for Windows. I updated build requirements for Windows in commit cf90f20eb2931285082fe2a5e1c7292e040c35cf 

I also pushed a new tarball for testing. Please test and report

closed 2016-08-02 by yoogx