OpenAADL/ocarina #76: Compilation fails

lounick, 2016-11-17:
Hi I'm trying to compile ocarina and I get the following error. My ADA knowledge is virtually 0 so I can't provide more help :(

```Making all in support
make[1]: Entering directory `/home/niko/src/ocarina/support'
make[1]: Nothing to be done for `all'.
make[1]: Leaving directory `/home/niko/src/ocarina/support'
Making all in tools
make[1]: Entering directory `/home/niko/src/ocarina/tools'
make[2]: Entering directory `/home/niko/src/ocarina/tools'
make[2]: Nothing to be done for `all-am'.
make[2]: Leaving directory `/home/niko/src/ocarina/tools'
make[1]: Leaving directory `/home/niko/src/ocarina/tools'
Making all in projects
make[1]: Entering directory `/home/niko/src/ocarina/projects'
if [ -f ../tools/mknodes/mknodes.gpr ]; then \
	   mkdir -p ../tools/mknodes/objects; \
	    \
	     gnatmake -P ../tools/mknodes/mknodes.gpr \
	     -XABS_SRC_DIR="/home/niko/src/ocarina/src" \
	     -XABS_BUILD_DIR="/home/niko/src/ocarina/src" \
	     -aP../projects -aP../projects \
	     -XBUILD=release ; \
	fi
gcc-4.6 -c -gnatec=/home/niko/src/ocarina/src/../projects/restrictions.adc -O2 -gnat12 -gnata -gnaty -gnatyu -gnatys -gnatpn -I- -gnatA /home/niko/src/ocarina/tools/mknodes/mknodes.adb
gcc-4.6 -c -gnatec=/home/niko/src/ocarina/src/../projects/restrictions.adc -O2 -gnat12 -gnata -gnaty -gnatyu -gnatys -gnatpn -I- -gnatA /home/niko/src/ocarina/tools/mknodes/parser.adb
gcc-4.6 -c -gnatec=/home/niko/src/ocarina/src/../projects/restrictions.adc -O2 -gnat12 -gnata -gnaty -gnatyu -gnatys -gnatpn -I- -gnatA /home/niko/src/ocarina/tools/mknodes/errors.adb
gcc-4.6 -c -gnatec=/home/niko/src/ocarina/src/../projects/restrictions.adc -O2 -gnat12 -gnata -gnaty -gnatyu -gnatys -gnatpn -I- -gnatA /home/niko/src/ocarina/tools/mknodes/lexer.adb
gcc-4.6 -c -gnatec=/home/niko/src/ocarina/src/../projects/restrictions.adc -O2 -gnat12 -gnata -gnaty -gnatyu -gnatys -gnatpn -I- -gnatA /home/niko/src/ocarina/tools/mknodes/namet.adb
gcc-4.6 -c -gnatec=/home/niko/src/ocarina/src/../projects/restrictions.adc -O2 -gnat12 -gnata -gnaty -gnatyu -gnatys -gnatpn -I- -gnatA /home/niko/src/ocarina/tools/mknodes/output.adb
gcc-4.6 -c -gnatec=/home/niko/src/ocarina/src/../projects/restrictions.adc -O2 -gnat12 -gnata -gnaty -gnatyu -gnatys -gnatpn -I- -gnatA /home/niko/src/ocarina/tools/mknodes/types.adb
gcc-4.6 -c -gnatec=/home/niko/src/ocarina/src/../projects/restrictions.adc -O2 -gnat12 -gnata -gnaty -gnatyu -gnatys -gnatpn -I- -gnatA /home/niko/src/ocarina/tools/mknodes/utils.adb
gcc-4.6 -c -gnatec=/home/niko/src/ocarina/src/../projects/restrictions.adc -O2 -gnat12 -gnata -gnaty -gnatyu -gnatys -gnatpn -I- -gnatA /home/niko/src/ocarina/tools/mknodes/locations.adb
gcc-4.6 -c -gnatec=/home/niko/src/ocarina/src/../projects/restrictions.adc -O2 -gnat12 -gnata -gnaty -gnatyu -gnatys -gnatpn -I- -gnatA /home/niko/src/ocarina/tools/mknodes/charset.adb
gnatbind -t -static -I- -x /home/niko/src/ocarina/tools/mknodes/objects/mknodes.ali
gnatlink /home/niko/src/ocarina/tools/mknodes/objects/mknodes.ali -o /home/niko/src/ocarina/tools/mknodes/mknodes
make 'GEN_MODE=soft' ../src/core/tree/ocarina-me_aadl-aadl_tree-nodes.ads ../src/core/tree/ocarina-me_aadl-aadl_instances-nodes.ads ../src/core/tree/ocarina-me_aadl_ba-ba_tree-nodes.ads ../src/core/tree/ocarina-me_real-real_tree-nodes.ads ../src/backends/ocarina-backends-ada_tree-nodes.ads ../src/backends/ocarina-backends-c_tree-nodes.ads ../src/backends/ocarina-backends-asn1_tree-nodes.ads ../src/backends/ocarina-backends-xml_tree-nodes.ads ../src/backends/ocarina-backends-pn-nodes.ads ../src/backends/ocarina-backends-mast_tree-nodes.ads
make[2]: Entering directory `/home/niko/src/ocarina/projects'
make[2]: `../src/core/tree/ocarina-me_aadl-aadl_tree-nodes.ads' is up to date.
make[2]: `../src/core/tree/ocarina-me_aadl-aadl_instances-nodes.ads' is up to date.
make[2]: `../src/core/tree/ocarina-me_aadl_ba-ba_tree-nodes.ads' is up to date.
make[2]: `../src/core/tree/ocarina-me_real-real_tree-nodes.ads' is up to date.
make[2]: `../src/backends/ocarina-backends-ada_tree-nodes.ads' is up to date.
make[2]: `../src/backends/ocarina-backends-c_tree-nodes.ads' is up to date.
make[2]: `../src/backends/ocarina-backends-asn1_tree-nodes.ads' is up to date.
make[2]: `../src/backends/ocarina-backends-xml_tree-nodes.ads' is up to date.
make[2]: `../src/backends/ocarina-backends-pn-nodes.ads' is up to date.
make[2]: `../src/backends/ocarina-backends-mast_tree-nodes.ads' is up to date.
make[2]: Leaving directory `/home/niko/src/ocarina/projects'
gnatmake \
	  -j4 -P ocarina-core.gpr -XBUILD=release -XLIBRARY_TYPE=static \
	  -XABS_SRC_DIR="/home/niko/src/ocarina/src" \
	  -XABS_BUILD_DIR="/home/niko/src/ocarina/src" \
	  -aP../projects -aP../projects;
gcc-4.6 -c -gnatec=/home/niko/src/ocarina/src/../projects/restrictions.adc -O2 -gnat12 -gnata -gnaty -gnatyu -gnatys -gnatpn -I- -gnatA /home/niko/src/ocarina/src/core/instance/ocarina-instances-components-features.adb
gcc-4.6 -c -gnatec=/home/niko/src/ocarina/src/../projects/restrictions.adc -O2 -gnat12 -gnata -gnaty -gnatyu -gnatys -gnatpn -I- -gnatA /home/niko/src/ocarina/src/core/instance/ocarina-instances-messages.adb
gcc-4.6 -c -gnatec=/home/niko/src/ocarina/src/../projects/restrictions.adc -O2 -gnat12 -gnata -gnaty -gnatyu -gnatys -gnatpn -I- -gnatA /home/niko/src/ocarina/src/core/instance/ocarina-instances-real_checker-queries-passing_predicates.adb
gcc-4.6 -c -gnatec=/home/niko/src/ocarina/src/../projects/restrictions.adc -O2 -gnat12 -gnata -gnaty -gnatyu -gnatys -gnatpn -I- -gnatA /home/niko/src/ocarina/src/core/model/ocarina-builder-aadl-components-flows.adb
gcc-4.6 -c -gnatec=/home/niko/src/ocarina/src/../projects/restrictions.adc -O2 -gnat12 -gnata -gnaty -gnatyu -gnatys -gnatpn -I- -gnatA /home/niko/src/ocarina/src/core/common_files/outfiles.adb
gcc-4.6 -c -gnatec=/home/niko/src/ocarina/src/../projects/restrictions.adc -O2 -gnat12 -gnata -gnaty -gnatyu -gnatys -gnatpn -I- -gnatA /home/niko/src/ocarina/src/core/model/ocarina-builder-aadl-components-modes.adb
gcc-4.6 -c -gnatec=/home/niko/src/ocarina/src/../projects/restrictions.adc -O2 -gnat12 -gnata -gnaty -gnatyu -gnatys -gnatpn -I- -gnatA /home/niko/src/ocarina/src/core/model/ocarina-builder-aadl_ba.ads
gcc-4.6 -c -gnatec=/home/niko/src/ocarina/src/../projects/restrictions.adc -O2 -gnat12 -gnata -gnaty -gnatyu -gnatys -gnatpn -I- -gnatA /home/niko/src/ocarina/src/core/instance/ocarina-instances-components-modes.adb
gcc-4.6 -c -gnatec=/home/niko/src/ocarina/src/../projects/restrictions.adc -O2 -gnat12 -gnata -gnaty -gnatyu -gnatys -gnatpn -I- -gnatA /home/niko/src/ocarina/src/core/model/ocarina-analyzer-aadl-finder.adb
gcc-4.6 -c -gnatec=/home/niko/src/ocarina/src/../projects/restrictions.adc -O2 -gnat12 -gnata -gnaty -gnatyu -gnatys -gnatpn -I- -gnatA /home/niko/src/ocarina/src/core/model/ocarina-builder-aadl-components-prototypes.adb
ocarina-aadl_values.ads:68:26: missing operand
gnatmake: "/home/niko/src/ocarina/src/core/model/ocarina-analyzer-aadl-finder.adb" compilation error
make[1]: *** [ocarina-core.gpr-build] Error 4
make[1]: Leaving directory `/home/niko/src/ocarina/projects'
make: *** [all-recursive] Error 1
```

[... 82 lines cut ...]
config.status: creating projects/ocarina.gpr
config.status: creating src/main/Makefile
config.status: creating src/config/ocarina-configuration.adb
config.status: creating src/frontends/Makefile
config.status: creating src/config/Makefile
config.status: creating src/core/Makefile
config.status: creating src/transfo/Makefile
config.status: creating src/backends/Makefile
config.status: creating src/python/Makefile
config.status: creating src/Makefile
config.status: creating support/Makefile
config.status: creating support/strip_wrapper.sh
config.status: creating tools/Makefile
config.status: creating tools/gendoc.py
config.status: creating tools/compare.py
config.status: creating ocarina-config
config.status: executing depfiles commands
configure: %%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%
configure: Ocarina configuration is complete!
configure: Including the following Ocarina modules: 
configure: GNU make command name: "make"
configure: %%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%
```

System is an Ubuntu 14.04 x86_64.

yoogx, 2016-11-17:
Hi,

I guess you used the default GNAT provided with Ubuntu 14.04
What is the output of "gnatls -v" in a terminal ?
It should be either GCC 6.1 or GNAT GPL 2016 (the one I test). Older versions may work. GCC 4.x is definitely too old as I require some extensions of Ada 2012

Regards

lounick, 2016-11-18:
Ok it is definitely the old version then. I'll try to get a newer one somehow and let you know.

Thanks for the fast responce! :smile: :+1:

closed 2016-11-18 by lounick