OpenAADL/ocarina #87: Ocarina bug detected

Hi-Zed, 2017-03-17:
Hello,
I'm having a problem when parsing my aadl file. At some point I get this error and I cannot pinpoint where is the problem.
I'm using Ocarina directly from Osate and I'm running the PolyORB-HI-C backend. I know that my model it's not compatible withe the backend, but I should be able to complete the instantiation.

+========================== OCARINA BUG DETECTED =========================+
| Detected exception: SYSTEM.ASSERTIONS.ASSERT_FAILURE                    |
| Error: ocarina-me_aadl-aadl_instances-entities.adb:142                  |
| Please refer to the User's Guide for more details.                      |
+=========================================================================+

raised SYSTEM.ASSERTIONS.ASSERT_FAILURE : ocarina-me_aadl-aadl_instances-entities.adb:142

Symbolic Traceback:

yoogx, 2017-03-17:
HI Luca, I need a model in order to investigate. Can you send it? Thanks

Hi-Zed, 2017-03-18:
This is the simplest model I can implement and get the error. It's a single process inside a system implementation.

Here the definition of the process:

````
package core
public
	data state_data	
	end state_data;
	
	data implementation state_data.impl
	end state_data.impl;
	
	thread main_loop
		features
			error_tr: out event port;
			run_tr: out event port;
			init_tr: out event port;
			close_tr: out event port;
			globalState: requires data access state_data.impl;
			notify_state: requires subprogram access state_manager;
	end main_loop;
	
	thread implementation main_loop.impl
		subcomponents
			prepare: subprogram function;
			spin: subprogram function;
			tear_down: subprogram function;
			error_handler: subprogram function;
		connections
			--transitions
			run_sig_pr: port prepare.signal -> run_tr ;
			error_sig_pr: port prepare.signal -> error_tr ;
			error_sig_sp: port spin.signal -> error_tr ;
			run_sig_sp: port spin.signal -> run_tr ;
			close_sig_er: port error_handler.signal -> close_tr ;
			run_sig_er: port error_handler.signal -> run_tr ;
			init_sig_er: port error_handler.signal -> init_tr ;
			
			--data access
			da1: data access  prepare.state_access -> globalState ;
			da2: data access  spin.state_access -> globalState ;
			da3: data access  tear_down.state_access -> globalState ;
			da4: data access  error_handler.state_access -> globalState ;
	end main_loop.impl;
	
	subprogram function
	features
		signal: out event port;
		state_access: requires data access state_data.impl;
	end function;
		
	subprogram state_manager
	end state_manager;
		
	process node
		features
			close: in event port;
			notify_state: requires subprogram access state_manager;
	end node;
	
	process implementation node.impl
		subcomponents
			main_thread: thread main_loop.impl in modes (init , running, closing, error);
			state: data state_data.impl;
	connections
			da: data access  state -> main_thread.globalState;
			sc: subprogram access main_thread.notify_state -> notify_state; 
	modes
			init: initial mode;
			running: mode;
			closing: mode;
			error: mode;
			running -[close]-> closing;
			running -[main_thread.run_tr]-> running;
			init -[close]-> closing;
			init -[main_thread.run_tr]-> running;
			init -[main_thread.error_tr]-> error;
			error -[main_thread.init_tr]-> init;
			error -[main_thread.run_tr]-> running;
			error -[main_thread.close_tr]-> closing;
	end node.impl;
end core;
````

And here the system implementation:

````
package test
public
	with core;
	
	system container
	end container;
	
	system implementation container.impl
		subcomponents
			test_pr: process core::node.impl;
	end container.impl;
end test;
````

yoogx, 2017-03-18:
Thanks for the reproducer. Problem is in Ocarina, when trying to resolve mode names main_thread.blah, i.e. mode of subcomponents. I'll try to address this as soon as time permits. In the mean time, you may comment out lines that uses main_thread.<mode_name> as triggers

closed 2017-04-07 by yoogx