OpenAADL/ocarina #88: Are prototypes supported?

Hi-Zed, 2017-03-23:
I'm trying to use prototypes to generalize the data associated with a port in a subcomponent, but I'm not sure if my AADL is wrong, or if Ocarina doesn't support them.
This is the example code I'm running.
```
package proto_test
public
	data message
	end message;
	
	process basic
		prototypes
			rd: data;
		features
			input: in data port rd;
			output: out data port message;
	end basic;
	
	process implementation basic.impl
	end basic.impl;
	
	system top			
	end top;
	
	system implementation top.impl
		subcomponents
			one: process basic.impl (rd => data message);
			two: process basic.impl (rd => data message);
		connections
			conn: port one.output -> two.input;
		
	end top.impl;
	
end proto_test;
```

The result I get after running ocarina is:
```
proto_test.aadl:25:04: one.output (entity reference)  and two.input (entity reference)  do not have compatible types
```

Hi-Zed, 2017-06-06:
Hello,
I'm trying again to use prototypes in my model and by doing this small modification:

```
package proto_test
public
	data message
	end message;
	
	process basic
		prototypes
			rd: data;
		features
			input: in data port rd;
			output: out data port rd;
	end basic;
	
	process implementation basic.impl
	end basic.impl;
	
	system top			
	end top;
	
	system implementation top.impl
		subcomponents
			one: process basic.impl (rd => data message);
			two: process basic.impl (rd => data message);
		connections
			conn: port one.output -> two.input;
		
	end top.impl;
	
end proto_test;
```

I set both ports as using the same prototypes, I get this error:

```
+========================== OCARINA BUG DETECTED =========================+
| Detected exception: SYSTEM.ASSERTIONS.ASSERT_FAILURE                    |
| Error: ocarina-me_aadl-aadl_tree-nodes.adb:789                          |
| Please refer to the User's Guide for more details.                      |
+=========================================================================+

raised SYSTEM.ASSERTIONS.ASSERT_FAILURE : ocarina-me_aadl-aadl_tree-nodes.adb:789
```

Hi-Zed, 2017-06-06:
Further investigations show that the problem may be that the prototype is not actually binded, and therefore "replaced" by the assigned type.

Seems like that the bug shown in my previous comment it's triggered by the fact that two ports with the same prototype have a compatible type, but this type cannot be analyzed further, since it's only a prototype.

yoogx, 2017-07-05:
After some investigation, I confirm Ocarina can parse prototypes, but not instantiate them. There has never been any demand in that direction. Furthermore, their status in AADL3 is being questioned. I'll decide on implementing them when we reach a consensus there

closed 2017-07-05 by yoogx