OpenAADL/ocarina #125: Detected exception: STORAGE_ERROR (Error: s-intman.adb:139 explicit raise)

vedranMv, 2017-11-29:
Hello!
    
    OCARINA VERSION: Ocarina 2017.x (Working Copy from r1deadbb)
    HOST MACHINE and OPERATING SYSTEM: Ubuntu 16.04.3 LTS (kernel 4.4.0-101-generic) 64-bit
    COMPILER VERSION: GNATLS 4.9.3, gcc version 5.4.0 20160609 (Ubuntu 5.4.0-6ubuntu1~16.04.5), OSATE2 (2.2.3.vupdate01)
    DESCRIPTION:
I'm trying to run PolyORB-HI-C code generation through Ocarina on an AADL project but I keep getting this error in console:

> Software.aadl:55:13: warning: faultD references a component type
> ocarina: Total: 0 error and 1 warning
> +========================== OCARINA BUG DETECTED =========================+
> | Detected exception: STORAGE_ERROR                                       |
> | Error: s-intman.adb:139 explicit raise                                  |
> | Please refer to the User's Guide for more details.                      |
> +=========================================================================+
> 
> Exception name: STORAGE_ERROR
> Message: s-intman.adb:139 explicit raise
> 
> Symbolic Traceback:

I'm not sure if it's ocarina error or problem with my model but in case of the later one, error message is not being very helpful :/
``faultD`` variable flagged as warning is defined as follows:
``faultD      : data Base_Types::Boolean{Data_Size => 1 Bytes;};``

    REPEAT BY:  Error is very consistent/repeatable on my platform. I have a model consisting of several .aadl files and includes both software & hardware. To create an error I select top-level system implementation from outline view then go Ocarina->Generate PolyORB-HI-C code (same result as going through OSATE->Ocarina bridge in toolstrip)

    SAMPLE FIX/WORKAROUND: None so far

Not sure if is helpful here but I'm able to run everything from 'Analyses' menu on an instance of my model, and I'm even able to generate MAST model by using Ocarina (OSATE->Ocarina bridge->Generate MAST model).

yoogx, 2017-11-29:
The warning on faultD is not the issue. Is it possible to share the model ? It would help me.
Also, since you compiled Ocarina from sources, please compile it again with debug activated.
( --enable-debug in configure ). That would provide more details on the location of the error.

vedranMv, 2017-11-29:
I've compiled Ocarina as you suggested and the error message now is different. What you also pointed me towards is that build_ocarina script used older version of gcc which I've deleted from my system and compiled everything with new version without building issues. Ocarina version is now different: ```Ocarina 2017.x (Working Copy from raa7075c)``` and bug seems to have disappeared but instead it says that it ```Cannot analyze AADL specifications```. 

I've also recompiled ocarina without ```--enable-debug``` flag and now with new version it produces the same output as the one compiled with the flag. 

This is what I get now:

```
/home/v125/Programming/osate/osateTools/ocarinaWS/build_debug/bin/ocarina -aadlv2 -g polyorb_hi_c -r Platform.impl -o (..list of all source files..)
aadl_project.aadl:108:03:  (constant property declaration) does not point to anything or to something unreachable
aadl_project.aadl:108:13:  (unique property type identifier) does not point to anything
aadl_project.aadl:108:30:  (signed aadlnumber)  is a unit literal but  (constant property declaration)  is not a unit type
deployment_properties.aadl:82:02: Allowed_Period (property definition declaration) does not point to anything
communication_properties.aadl:48:09:  (unique property type identifier) does not point to anything
communication_properties.aadl:44:69:  (signed aadlnumber)  is a unit literal but  (property type)  is not a unit type
communication_properties.aadl:48:09:  (unique property type identifier) does not point to anything
communication_properties.aadl:44:77:  (signed aadlnumber)  is a unit literal but  (property type)  is not a unit type
communication_properties.aadl:48:09:  (unique property type identifier) does not point to anything
communication_properties.aadl:57:72:  (signed aadlnumber)  is a unit literal but  (property type)  is not a unit type
communication_properties.aadl:48:09:  (unique property type identifier) does not point to anything
communication_properties.aadl:57:80:  (signed aadlnumber)  is a unit literal but  (property type)  is not a unit type
communication_properties.aadl:70:04:  (record type element) does not point to anything
communication_properties.aadl:74:02: Actual_Latency (property definition declaration) does not point to anything
communication_properties.aadl:77:02: Latency (property definition declaration) does not point to anything
communication_properties.aadl:48:09:  (unique property type identifier) does not point to anything
memory_properties.aadl:19:44:  (signed aadlnumber)  is a unit literal but  (record type element)  is not a unit type
communication_properties.aadl:48:09:  (unique property type identifier) does not point to anything
memory_properties.aadl:19:52:  (signed aadlnumber)  is a unit literal but  (record type element)  is not a unit type
memory_properties.aadl:27:03:  (record type element) does not point to anything
memory_properties.aadl:38:03:  (record type element) does not point to anything
memory_properties.aadl:78:03:  (record type element) does not point to anything
ARINC653.aadl:5:01: Module_Major_Frame (property definition declaration) does not point to anything
ARINC653.aadl:9:01: Sampling_Refresh_Period (property definition declaration) does not point to anything
ARINC653.aadl:24:01: Timeout (property definition declaration) does not point to anything
ARINC653.aadl:136:01: Time_Capacity (property definition declaration) does not point to anything
SEI.aadl:38:03: Partition_Latency (property definition declaration) does not point to anything
Timing_Properties.aadl:7:02: Activate_Deadline (property definition declaration) does not point to anything
Timing_Properties.aadl:10:02: Activate_Execution_Time (property definition declaration) does not point to anything
Timing_Properties.aadl:13:02: Compute_Deadline (property definition declaration) does not point to anything
Timing_Properties.aadl:16:02: Compute_Execution_Time (property definition declaration) does not point to anything
Timing_Properties.aadl:19:02: Client_Subprogram_Execution_Time (property definition declaration) does not point to anything
Timing_Properties.aadl:22:02: Deactivate_Dealing (property definition declaration) does not point to anything
Timing_Properties.aadl:25:02: Deactivate_Execution_Time (property definition declaration) does not point to anything
Timing_Properties.aadl:28:02: Deadline (property definition declaration) does not point to anything
Timing_Properties.aadl:31:02: First_Dispatch_Time (property definition declaration) does not point to anything
Timing_Properties.aadl:34:02: Dispatch_Jitter (property definition declaration) does not point to anything
Timing_Properties.aadl:37:02: Dispatch_Offset (property definition declaration) does not point to anything
Timing_Properties.aadl:40:02: Execution_Time (property definition declaration) does not point to anything
Timing_Properties.aadl:43:02: Finalize_Deadline (property definition declaration) does not point to anything
Timing_Properties.aadl:46:02: Finalize_Execution_Time (property definition declaration) does not point to anything
Timing_Properties.aadl:49:02: Initialize_Deadline (property definition declaration) does not point to anything
Timing_Properties.aadl:52:02: Initialize_Execution_Time (property definition declaration) does not point to anything
Timing_Properties.aadl:55:02: Load_Deadline (property definition declaration) does not point to anything
[... cut ...]
Timing_Properties.aadl:102:02: Scheduler_Quantum (property definition declaration) does not point to anything
Timing_Properties.aadl:105:02: Thread_Swap_Execution_Time (property definition declaration) does not point to anything
Timing_Properties.aadl:108:02: Frame_Period (property definition declaration) does not point to anything
Timing_Properties.aadl:111:02: Slot_Time (property definition declaration) does not point to anything
EMV2.aadl:148:02:  (record type element) does not point to anything
Cannot analyze AADL specifications
 ```

yoogx, 2017-11-29:
Is it possible to share the model ? it would ease the analysis on my end. thanks

vedranMv, 2017-11-29:
OSATE2 (2.2.3.vupdate01), working on sending you the project over email.

yoogx, 2017-11-30:
After analysis of the model, this is not a bug itself, just a lack of diagnosis. Closing to discuss this in private

closed 2017-11-30 by yoogx