OpenAADL/ocarina #172: OCARINA BUG DETECTED: STORAGE_ERROE => s-intman.adb:136 explicit raise

parvanehAghajani, 2018-08-23:
OCARINA VERSION:
     Ocarina 2017.x (Working Copy from rcdc5d06)
     Copyright (c) 2003-2009 Telecom ParisTech, 2010-2018 ESA & ISAE
HOST MACHINE and OPERATING SYSTEM:
     Linux ubuntu 4.15.0-20-generic #21-Ubuntu SMP Tue Apr 24 06:16:15 UTC 2018 x86_64 x86_64         
      x86_64 GNU/Linux
COMPILER VERSION
     GNATLS 7.3.0
     Copyright (C) 1997-2017, Free Software Foundation, Inc.

    DESCRIPTION:
Hello. 
I'm trying to use ocarina to generate a mast model from an AADL file on OSATE 2.3.4.vfinal but I get this error: 

ocarina: Cannot find a root system
+========================== OCARINA BUG DETECTED =========================+
| Detected exception: STORAGE_ERROR                               |
| Error: s-intman.adb:136 explicit raise                                  |
| Please refer to the User's Guide for more details.              |
+=========================================================================+
raised STORAGE_ERROR : s-intman.adb:136 explicit raise.

I'm not sure if this error is related to the addl file (like correct properties for components) or not, but I tried to generate a MAST model for simpler AADL files and it works. By calling other options like "generate statistic", "generate petri-net" and ... I get the same error. I should point out that I'm using Ubuntu VM on Mac. 
Do you think that this issue could be related to memory and is there any option for increasing memory for performing code generation?

parvanehAghajani, 2018-08-23:
Thank you for your quick response :) 
I removed the system features and it works now.

yoogx, 2018-08-23:
Perfect. I close the issue

closed 2018-08-23 by yoogx