OpenAADL/ocarina #173: Segmentation fault after running Ocarina compiled executable

svenTeum, 2018-09-17:
OCARINA VERSION:
Ocarina 2017.x (Working Copy from rf4cbc2d)
Copyright (c) 2003-2009 Telecom ParisTech, 2010-2018 ESA & ISAE
Build date: 

HOST MACHINE and OPERATING SYSTEM:
Linux ines 4.9.0-6-amd64 #1 SMP Debian 4.9.88-1+deb9u1 (2018-05-07) x86_64 GNU/Linux
    
COMPILER VERSION
gcc (Debian 6.3.0-18+deb9u1) 6.3.0 20170516

DESCRIPTION:
After compiling AADL code with
`ocarina -aadlv2 -y -b -g polyorb_hi_c xx.aadl`
I get an executable that I can run but I get a segmentation fault error and result.csv file is 0 bytes.
```
./xx_sys_impl/actuator/actuator
soon monitoring
 stdout            | info    | The initialization finished successfully without homotopy method.
stdout            | info    | The initialization finished successfully without homotopy method.
stdout            | info    | The initialization finished successfully without homotopy method.
stdout            | info    | The initialization finished successfully without homotopy method.
stdout            | info    | The initialization finished successfully without homotopy method.
Segmentation fault
```
This model contains 5 FMUs generated in OpenModelica. Unfortunately the full model can't be shared so is there any way that I can debug that error?

Thanks

yoogx, 2018-09-17:
Can you compile with debug activated and use gdb to know where the error is ? It may either be in the FMU wrapper itself, or PolyORB-HI/C. I suppose it is the FMU wrapper itself, but cannot guess without further details.

svenTeum, 2018-09-17:
This is the gdb output after compiling with -d flag. I'm not sure if it gives useful information.
```
gdb ./xx_sys_impl/actuator/actuator
GNU gdb (Debian 7.12-6) 7.12.0.20161007-git
Copyright (C) 2016 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.  Type "show copying"
and "show warranty" for details.
This GDB was configured as "x86_64-linux-gnu".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
<http://www.gnu.org/software/gdb/documentation/>.
For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from ./xx_sys_impl/actuator/actuator...done.
(gdb) start
Temporary breakpoint 1 at 0x9964: file main.c, line 36.
Starting program: xx_sys_impl/actuator/actuator 

[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".

Temporary breakpoint 1, main () at main.c:36
warning: Source file is more recent than executable.
36	  __po_hi_initialize_early ();
(gdb) 
(gdb) continue
Continuing.
soon monitoring
 stdout            | info    | The initialization finished successfully without homotopy method.
stdout            | info    | The initialization finished successfully without homotopy method.
stdout            | info    | The initialization finished successfully without homotopy method.
stdout            | info    | The initialization finished successfully without homotopy method.
stdout            | info    | The initialization finished successfully without homotopy method.
[New Thread 0x7ffff59a5700 (LWP 4267)]
[New Thread 0x7ffff51a4700 (LWP 4268)]
[New Thread 0x7ffff49a3700 (LWP 4269)]
[New Thread 0x7ffff41a2700 (LWP 4270)]
[New Thread 0x7ffff39a1700 (LWP 4271)]

Thread 4 "actuator" received signal SIGSEGV, Segmentation fault.
[Switching to Thread 0x7ffff49a3700 (LWP 4269)]
0x0000000000000000 in ?? ()
```

svenTeum, 2018-09-19:
Have you tested any FMUs generated with OpenModelica?

yoogx, 2018-09-19:
Cannot remember from the top of my head wether we tested OpenModelica or JModelica
openaadl/fmu2aadl repository has the examples we tested

yoogx, 2020-01-03:
The following patch addresses the issue when using FMUs generated by OpenModelica

closed 2020-01-03 by yoogx