OpenAADL/ocarina #184: Issue about generating polyorb_hi_ada code on Osate

sinkinben, 2019-04-16:
A very simple AADL system.
```
package test
public
	system sin	
	end sin;
	system implementation sin.impl	
	end sin.impl;
end test;
```
I select `System Impl sin.impl`, right click, and select `Ocarina->PolyORB-HI-Ada Code`.
PS: The `PolyORB-HI-C Code` can work.
Osate console output:
```
E:\ProgramFiles\Git\usr\local\bin\ocarina.exe -aadlv2 -g polyorb_hi_ada -r sin.impl D:\programfiles\osate-64bit(2.2.3)\workspace\aadlVerification\AADL_ZBL\test.aadl
Backends: warning : sin_impl already exists
Backends: fatal error : E:\ProgramFiles\Git\usr\local\include\ocarina\runtime\polyorb-hi-ada is not a valid runtime directory
```

The tree structure of dir `E:\ProgramFiles\Git\usr\local\include\ocarina\runtime\`
```
├─aadl_xml
├─alloy
│  ├─contract
│  │  ├─analysis
│  │  └─goal
│  ├─lib
│  └─meta
├─cheddar
├─lnt
├─polyorb-hi-c
│  ├─include
│  │  ├─drivers
│  │  │  └─configuration
│  │  └─lua
│  ├─libs
│  │  ├─leon-rtems4.11
│  │  ├─leon-rtems4.8
│  │  ├─linux32
│  │  └─linux64
│  ├─make
│  ├─src
│  │  └─drivers
│  │      └─configuration
│  └─utils
└─python
```
Is it missing something when I install ocarina?

yoogx, 2019-04-16:
Yes, the PolyORB-HI/Ada has not been installed. How did you install Ocarina?
Very old releases did not include it for Windows

sinkinben, 2019-04-17:
I git clone the repo [ocarina-build](https://github.com/OpenAADL/ocarina-build)
And exec the command
```
./build_ocarina.sh --scenario=fresh-install --prefix=$PWD/ocarina_install
```

But there is something wrong. So I follow the tutorial [Ocarina Docs](https://ocarina.readthedocs.io/en/latest/installation.html)
Use these commands to build:
```
./build_ocarina.sh -b
make
make install
```

sinkinben, 2019-04-17:
I rebuild ocarina. Now, the `Generate Polyorb-Ada-HI Code` can work correctly.
I close this issue.
Thanks: )

closed 2019-04-17 by sinkinben

yoogx, 2019-04-17:
Can you tell me what was the issue in the build script ? both lines should be equivalent in what you did.

sinkinben, 2019-04-17:
That's caused by Git-for-Windows.  The operation `git clone` in the git-bash is very slow.
Like this:
```
$ ./build_ocarina.sh --scenario=fresh-install --prefix=$PWD/ocarina_install
OS:        MINGW64_NT-10.0 2.9.0(0.318/5/3) x86_64
Compiler:  GNATMAKE Community 2018 (20180523-73)
autoconf:  autoconf (GNU Autoconf) 2.69
automake:  automake (GNU automake) 1.16.1
[2019-04-17-16:36] Checkout the Ocarina sources[2019-04-17-16:36] Checkout the Ocarina sources         [FAILED]
Cloning into 'ocarina'...
fatal: unable to access 'https://github.com/OpenAADL/ocarina.git/': Failed to connect to github.com port 443: Timed out
```
So I download `aadlib`, `polyorb-hi-c` and `polyorb-hi-ada` by Explorer, execute the three commands described above.