OpenAADL/ocarina #186: Python bindings build error

imadari, 2019-04-30:
OCARINA VERSION:
```
Ocarina 2017.x (Working Copy from r34f5219)
Copyright (c) 2003-2009 Telecom ParisTech, 2010-2019 ESA & ISAE
Build date: 
```

HOST MACHINE and OPERATING SYSTEM:
```
Linux ginkas 4.15.0-48-generic #51-Ubuntu SMP Wed Apr 3 08:28:49 UTC 2019 x86_64 x86_64 x86_64 GNU/Linux
```

COMPILER VERSION
```
GNATLS 7.3.0
Copyright (C) 1997-2017, Free Software Foundation, Inc.

Source Search Path:
   <Current_Directory>
   /usr/lib/gcc/x86_64-linux-gnu/7/adainclude

Object Search Path:
   <Current_Directory>
   /usr/lib/gcc/x86_64-linux-gnu/7/adalib

Project Search Path:
   <Current_Directory>
   /usr/x86_64-linux-gnu/lib/gnat
   /usr/x86_64-linux-gnu/share/gpr
   /usr/share/gpr
   /usr/lib/gnat
 ```

DESCRIPTION:
From ocarina-build directory:

```
./build_ocarina.sh --scenario=fresh-install --prefix=<myhomedir>/work/ocarina_install --build-info --enable-debug --enable-python
```

Error message:
```
ocarina-python.adb:72:07: no candidate interpretations match the actuals:
ocarina-python.adb:72:07: missing argument for parameter "Data" in call to "Set" declared at gnatcoll-scripts.ads:917
ocarina-python.adb:72:07: too many arguments in call to "Set"
```

yoogx, 2019-05-01:
Can you tell me which version of GNATColl you used ?

imadari, 2019-05-01:
Sure!

```sh
dpkg -l | grep libgnat
```

```sh
libgnat-7:amd64                            7.3.0-27ubuntu1~18.04 
libgnatcoll-python16.1.0:amd64             17.0.2017-3                                 
libgnatcoll-python16.1.0-dev               17.0.2017-3                           
libgnatcoll16.1.0:amd64                    17.0.2017-3                 
libgnatcoll16.1.0-dev                      17.0.2017-3
libgnatvsn7:amd64                          7.3.0-27ubuntu1~18.04                 
```

jjhugues, 2020-12-02:
The Python binding is now know to compile, see 
https://github.com/OpenAADL/ocarina/commit/0c00cdb53e12e371aeae1f7ad5a40ac7ca53f934

and the latest master

closed 2020-12-02 by jjhugues