OpenAADL/ocarina #224: Exit if no root system present

yoogx, 2019-08-02:
If no root system is found, Ocarina should exit immediately. Otherwise, an exception is raised.

closed 2019-08-02 by yoogx

maxime-esa, 2019-08-03:
However, it's a bit rude for the caller to exit the application like that. You should return No_System instead.

The calling side is already expecting this case:
```
   Ocarina.Options.Root_System_Name :=
        Get_String_Name ("taste_dataview.others");

      if DV_Root /= No_Node then
         System := Root_System (Instantiate_Model (Root => DV_Root));  -- Here you should not exit!
      end if;

      if No (System) then
         raise Data_View_Error with "Could not instantiate Data View";
      end if;
```

yoogx, 2019-08-05:
The bug report asked to exit immediately ;-)

I adjusted the logic to either exit or return no_node

In your code above, check the return value of Instantiate_Model, we may return no_node, and thus the call to Root_System could raise an exception

maxime-esa, 2019-08-06:
Alright, perfect now. Thanks!