OpenAADL/ocarina #237: Does Ocarina code generation tool covers Blackboard and Semaphores ?

CaptainSeum, 2019-10-08:
Hi,

I tried to build an example to test code generation capabilities regarding the annex Arinc653. It works well for pretty much everything, but it does not seems to pick up blackboards and semaphores. I followed the instructions described in the AADL annex Arinc653 to specify those component.

Does Ocarina covers those concepts ? is there a specific way to specify blackboards and semaphores for Ocarina to parse them ?

Thanks

yoogx, 2019-10-08:
your model might be incomplete, a quick examination of Ocarina source code demonstrates we emit code for blackboards and semaphores

closed 2019-10-08 by yoogx

CaptainSeum, 2019-10-08:
Ok, is there any model example that contains semaphores/blackboards that works with Ocarina ?

yoogx, 2019-10-08:
ADIRU is using blackboard
POK may have old models with semaphores