OpenAADL/ocarina #247: STORAGE_ERROR exception on every invocation

dordow, 2020-02-10:
OCARINA VERSION: 

2017.1

    HOST MACHINE and OPERATING SYSTEM: 

5.5.2-arch1-1 #1 SMP PREEMPT Tue, 04 Feb 2020 18:56:18 +0000 x86_64 GNU/Linux 
Asus ZenBook

    COMPILER VERSION
9.2.0

    DESCRIPTION:
I installed ocarina on Arch Linux (5.5.2-arch1-1) using the
semi-automatic build. The build script passed so I assume that I have
got a correct ocarina installation. However, whenever I call ocarina
with any option and any input, I get a STORAGE_ERROR exception raised
by s-intman.adb but without any further information:

`+========================== OCARINA BUG DETECTED =========================+
| Detected exception: STORAGE_ERROR                                       |
| Error: s-intman.adb:136 explicit raise                                  |
| Please refer to the User's Guide for more details.                      |
+=========================================================================+

raised STORAGE_ERROR : s-intman.adb:136 explicit raise

Symbolic Traceback:
`

This is similar to previously reported issues but not identical since
it happens on every invocation (except ocarina -h).
Thanks in advance for assistance

yoogx, 2020-02-10:
I tried building a Dockerfile for arch linux; following the templates from https://github.com/openaadl/ocarina-docker

How did you install gprbuild? the package seems missing

dordow, 2020-02-10:
I first installed gprbuild-bootstrap-git from AUR (gprbuild-bootstrap from AUR is out of date and broken), and then I downloaded and run 
`build_ocarina.sh --scenario=fresh-install --prefix=<local directory>`
That reported a couple of PASSED, no warning or fail. 
`ocarina -h` 
works, but nothing else.
Looking into gprbuild as you have suggested:  one needs 
- gprbuild-git which depends on 
- libgpr(-git ) which depends on
- libgpr(-git) which depends on 
- xmlada(-git) 
where makepkg fails with errors:
`gprconfig: can't find a native toolchain for language 'ada'`
However, 
`pacman -Qi gcc-ada`
shows that gcc-ada is installed. Is there anything else needed?

Btw, thanks for your fast response.

yoogx, 2020-02-11:
I will try later with this gpbuild. In the mean time; I suggest you try with GNAT GPL 2019. It is the one I use for my own tests.

dordow, 2020-02-12:
Do you mean by GNAT GPL 2019 the [gnat-gps](https://aur.archlinux.org/packages/gnat-gps/) AUR package? Or a direkt download from AdaCore. The former is also outdated and has problems.

yoogx, 2020-02-12:
I meant GNAT GPL 2019 from libre.adacore.com.

If you check https://github.com/OpenAADL/ocarina/blob/master/.travis.yml you will see it is the one I use for reg tests. My experience is that except for specific distributions like debian, the support for GNAT is somewhat fragile.

dordow, 2020-02-14:
I have installed GNAT following the instructions of AdaCore, then tried to build ocarina using the information that you have provided:
```

./build_ocarina.sh  --scenario=travis-ci --prefix=~/bin/ocarina
Guessing OS
linux
OS:        Arch Linux x86_64
Compiler:  GNATMAKE 9.2.0
autoconf:  autoconf (GNU Autoconf) 2.69
automake:  automake (GNU automake) 1.16.1
[2020-02-15-00:01] Checkout the Ocarina sources                                                                 [PASSED]
[2020-02-15-00:01] Fetching runtimes 'polyorb-hi-ada polyorb-hi-c aadlib'                                       [PASSED]
[2020-02-15-00:01] Reconfiguring (Ocarina)                                                                      [PASSED]
[2020-02-15-00:01] First configure (Ocarina)                                                                    [PASSED]
[2020-02-15-00:01] Doing 'make' (Ocarina)                                                                       [PASSED]
[2020-02-15-00:02] Doing 'make install' (Ocarina)                                                               [PASSED]
Ocarina installation is now complete !
Note: you may want to add
<<snip>>

[2020-02-15-00:02] Testing (Ocarina)[2020-02-15-00:03] Testing (Ocarina)                                                                            [FAILED]

<<snip>>

2020-02-14-22:17] Testing (Ocarina)[2020-02-14-22:18] Testing (Ocarina)                                                                            [FAILED]

<< snip lots of Ada code >>

---------------------------------------------------------------------------
FAILURES : 327 / 327
---------------------------------------------------------------------------
Result trace is stored in /home/xxx/install/ocarina-build/ocarina/log.wUDTL0
make[3]: *** [Makefile:831: test-suite-internal] Error 1
make[3]: Leaving directory '/home/xxx/install/ocarina-build/ocarina'
make[2]: *** [Makefile:867: check-local] Error 2
make[2]: Leaving directory '/home/xxx/install/ocarina-build/ocarina'
make[1]: *** [Makefile:702: check-am] Error 2
make[1]: Leaving directory '/home/xxx/install/ocarina-build/ocarina'
make: *** [Makefile:420: check-recursive] Error 1

```
Interestingly: ocarina did not throw an exception in the tests. But calling ocarina gives
a slightly different error now: 

```
ocarina --version
+========================== OCARINA BUG DETECTED =========================+
| Detected exception: CONSTRAINT_ERROR                                    |
| Error: ocarina-configuration.adb:188 access check failed                |
| Please refer to the User's Guide for more details.                      |
+=========================================================================+

raised CONSTRAINT_ERROR : ocarina-configuration.adb:188 access check failed
Call stack traceback locations:
0x55775c8d5227 0x55775c8bec14 0x55775c8c58b0 0x7f7bcba88151 0x55775c8b930c 0xfffffffffffffffe

Symbolic Traceback:
0x55775c8d5227
0x55775c8bec14
0x55775c8c58b0
[/usr/lib/libc.so.6]
0x7f7bcba88151
0x55775c8b930c
0xfffffffffffffffe

```

yoogx, 2020-02-14:
Please check your PATH; you are still using GCC 9.2.0
Here is my output

neraka% ./build_ocarina.sh  --scenario=travis-ci 
Guessing OS
osx
OS:        Darwin 18.7.0 x86_64
Compiler:  GNATMAKE Community 2019 (20190517-83)
autoconf:  autoconf (GNU Autoconf) 2.69

and travis output https://travis-ci.org/OpenAADL/ocarina/jobs/634548512

yoogx, 2020-02-14:
Your exception trace is interesting. Can you please confirm that Ocarina is in your path?
Line 188 is

         declare
            Dir : constant String
              := Get_Install_Dir (Locate_Exec_On_Path (Command_Name).all);

Locate_Exec_On_Path fails to find ocarina in the PATH

dordow, 2020-02-16:
Checking paths is a good hint, I have got some strange behaviour. When calling
`which ocarina`
I get 
`/home/frank/bin/ocarina/bin/ocarina`
Executing 
`ocarina --version`
runs into the exception. Calling
`/home/frank/bin/ocarina/bin/ocarina --version`
works! I have used 
`./build_ocarina.sh  --scenario=travis-ci --prefix=~/bin/ocarina`
because my build directory is different from the installation directory. I will try to build ocarina in the target directory without the prefix option.

yoogx, 2020-02-17:
The prefix should not have an impact. Out of curiosity, which shell are you using?

dordow, 2020-02-17:
I use bash and PATH and LD_LIBRARY_PATH are (re-)defined in .bashrc.

dordow, 2020-02-20:
I have ocarina installed on Windows 10 over cygwin using ocarina-build, no problems encountered. There are older versions of gcc and GNATMAKE, both are 7.4.0. So it seems that the Ada ecosystem on Arch is currently a bit inconsistent. Are there plans to provide a docker for Archlinux?

yoogx, 2020-02-20:
Good to know it works on Windows. You should be able to generate code targeting the Native (POSIX) target. I barely tested it so feel free to ask

There are docker images here: https://github.com/openaadl/ocarina-docker
You can use any of them on top of your ArchLinux installation; I recommend the centos or debian one

yoogx, 2020-03-06:
This is not an Ocarina bug; but a limit in ArchLinux, recommendation is to install GNAT CE 2019

closed 2020-03-06 by yoogx