OpenAADL/ocarina #250: Distributing package definitions over separate files

dordow, 2020-03-14:
OCARINA VERSION:
Ocarina 2017.x (Working Copy from rd018610)
Copyright (c) 2003-2009 Telecom ParisTech, 2010-2020 ESA & ISAE
Build date: 

    HOST MACHINE and OPERATING SYSTEM:
Arch Linux 5.5.8-arch1-1

    COMPILER VERSION
n/a

    DESCRIPTION:
I am new to AADL and to ocarina so this question may be stupid but I could not find a solution yet. I have defined a one package with a data definition in a file named data_defs.aadl, with the following
content:
```
package data_defs
public
  data mydata
  end mydata;

  data implementation mydata.impl
  end mydata.impl;
end data_defs;
```
In a separate file I define a system with a reference to that data definition:
```
package mysystem
public
  with data_defs;

  system mysystem
  features
    inport : in data port data_defs::mydata;
  end mysystem;

  system implementation mysystem.impl
  end mysystem.impl;
end mysystem;
```
When I run ocarina with
`ocarina -vy -aadlv1 -p mysystem.aadl
`
I get the following error message:
```
mysystem.aadl:8:15: parsing Package_Specification, Empty packages are not allowed
mysystem.aadl:12:01: parsing AADL_Declaration, unexpected keyword 'end'
Cannot parse AADL specifications
```
When I run ocarina with AADL v2, it parses the specification. When I put all definitions in a single file, it also works with V1. What is my mistake? Is it not possible to use separate files with V1?
Thanks in advance
Frank

yoogx, 2020-03-14:
AADLv1 is the old version of the standard. v2 adds a couple of new keywords, one is "with" + true namespace. I would recommend you stick to v2, and uses OSATE as IDE to have quickfixes for this type of syntax errors.

closed 2020-03-16 by yoogx

dordow, 2020-03-18:
Thank you for your explanation. It seems that v1 requires to have the full model in a single file. However, I will follow your advice and change to v2. 
I prefer command line tools to Java/Eclipse for various reasons so ocarina fits my conditions better than OSATE.

yoogx, 2020-03-18:
Sure. I also prefer emacs + text editor.You may want to check this companion project:
https://github.com/OpenAADL/AADLib/

It has (basic) modes for vim and emacs + collection of examples