OpenAADL/ocarina #252: OCARINA BUG DETECTED: SYSTEM.ASSERTIONS.ASSERT_FAILURE ocarina-me_aadl-aadl_instances-entities.adb:142

ghostezhouxingyu, 2020-03-20:
OCARINA VERSION:
Ocarina 2017.1 (Working Copy from rfb77e27)
Copyright (c) 2003-2009 Telecom ParisTech, 2010-2017 ESA & ISAE
 
HOST MACHINE and OPERATING SYSTEM:
Linux waker-virtual-machine 4.15.0-91-generic #92~16.04.1-Ubuntu SMP Fri Feb 28 14:57:22 UTC 2020 x86_64 x86_64 x86_64 GNU/Linux

DESCRIPTION:
+========================== OCARINA BUG DETECTED =========================+
| Detected exception: SYSTEM.ASSERTIONS.ASSERT_FAILURE                    |
| Error: ocarina-me_aadl-aadl_instances-entities.adb:142                  |
| Please refer to the User's Guide for more details.                      |
+=========================================================================+
raised SYSTEM.ASSERTIONS.ASSERT_FAILURE : ocarina-me_aadl-aadl_instances-entities.adb:142

Symbolic Traceback:

I'm using ocarina to generate xml file from an AADL file, here's my source code
   
package basenode
public
	with Data_Model;
	
	subprogram notify_state
	end notify_state;
	--
	subprogram implementation notify_state.impl
	end notify_state.impl;
	--
	process RosNode
		features
			close:in event port;
			notify_state:provides subprogram access notify_state.impl;
	end RosNode;
	--
	data internal_state
		properties
    		Data_Model::Data_Representation => integer;
	end internal_state;
	--
	data implementation internal_state.impl
	end internal_state.impl;
	--
	thread main_thread
		features
			internal_state:requires data access internal_state;
			error:out event port;
			run:out event port;
			init:out event port;
			close:out event port;
			notify_state:provides subprogram access notify_state.impl;
	end main_thread;
	--
	subprogram prepare
		features
			prepare_dc:requires data access internal_state;
			prepare_event:out event port;
	end prepare;
	--
	subprogram spin
		features
			spin_dc:requires data access internal_state;
			spin_event:out event port;
	end spin;
	--
	subprogram tear_down
		features
			tear_down_dc:requires data access internal_state;
			tear_down_event:out event port;
	end tear_down;
	--
	subprogram error_handler
		features
			error_handler_dc:requires data access internal_state;
			error_handler_event:out event port;
	end error_handler;
	--
	thread implementation main_thread.impl
		calls main:{
			prepare:subprogram prepare;
			spin:subprogram spin;
			tear_down:subprogram tear_down;
			error_handler:subprogram error_handler;
		};
		connections
			main_to_prepare:data access internal_state->prepare.prepare_dc;
			main_to_spin:data access internal_state->spin.spin_dc;
			main_to_tear_down:data access internal_state->tear_down.tear_down_dc;
			main_to_error_handler:data access internal_state->error_handler.error_handler_dc;
			prepare_to_error:port prepare.prepare_event->error;
			prepare_to_run:port prepare.prepare_event->run;
			spin_to_error:port spin.spin_event->error;
			spin_to_run:port spin.spin_event->run;
			error_handler_to_run:port error_handler.error_handler_event->run;
			error_handler_to_init:port error_handler.error_handler_event->init;
			error_handler_to_close:port error_handler.error_handler_event->close;
	
	end main_thread.impl;
	
	process implementation RosNode.base
		subcomponents
			main_thread:thread main_thread.impl;
			internal_state_data:data internal_state.impl;
		connections
			is_to_main:data access internal_state_data->main_thread.internal_state;
		
		modes
			init:initial mode;
			running:mode;
			error:mode;
			closing:mode;
			init -[main_thread.error]->error;
			init -[main_thread.run]->running;
			init -[close]->closing;
			running -[main_thread.run]->running;
			running -[main_thread.error]->error;
			running -[close]->closing;
			error -[main_thread.init]->init;
			error -[main_thread.run]->running;
			error -[main_thread.close]->closing;
	end RosNode.base;
	--
	system ros
	end ros;
	--
	system implementation ros.impl
		subcomponents 
			node: process RosNode.base;
	end ros.impl;
end basenode;

yoogx, 2020-03-20:
You should update and compile from sources.
Remove the "modes" part, it is the source of the crash. If I do, I get the following errors
➜  github ocarina -aadlv2 -g aadl_xml -f t.aadl
t.aadl:60:02 Backends: fatal error : Multiple subprogram calls in the thread call sequence are not supported. You should encapsulate them in a wrapper subprogram.

ghostezhouxingyu, 2020-03-21:
> You should update and compile from sources.
> Remove the "modes" part, it is the source of the crash. If I do, I get the following errors
> ➜ github ocarina -aadlv2 -g aadl_xml -f t.aadl
> t.aadl:60:02 [...]

Thanks.
Btw, when it happens like this:
+========================== OCARINA BUG DETECTED =========================+
| Detected exception: SYSTEM.ASSERTIONS.ASSERT_FAILURE                    |
| Error: ocarina-me_aadl-aadl_instances-nodes.adb:187                     |
| Please refer to the User's Guide for more details.                      |
+=========================================================================+

raised SYSTEM.ASSERTIONS.ASSERT_FAILURE : ocarina-me_aadl-aadl_instances-nodes.adb:187

Symbolic Traceback:

is it because i used "event port" in my aadl file?

jjhugues, 2020-12-02:
Please provide a complete example I can test.

closed 2020-12-02 by jjhugues