OpenAADL/ocarina #261: Unable to get IResource for Resource when running

nuaaysh, 2020-06-28:
Dear Authors of Ocarina,
     I have two questions about the Ocarina:
     Firstly, Where can I find the details about how your tool generates PolyORB-HI/Ada code from AADL models? I only find the PolyORB-HI/C part in the User’s Guide. 
   
    Secondly, as the title said I get an error message when I try to use Ocarina to generate Ada code,
     [image]

    The test example is from your example https://github.com/OpenAADL/polyorb-hi-ada/tree/master/examples/aadlv2/prime_numbers

some information is listed as follows:
OS Ubuntu20, x64
OSATE2: 2.7.1.vfinal
OSATE2-Ocarina plugin 0.0.1.201302151532
Ocarina 2017.x (Working Copy from ree7273d)
Copyright (c) 2003-2009 Telecom ParisTech, 2010-2020 ESA & ISAE
Build date: Jun 27 2020 18:59:47

Many Thanks!

yoogx, 2020-06-28:
PolyORB-HI/Ada has its own doc that you can build separately.
The second part should have been addressed, the plugin version you use does not make sense. OSATE  has a specific mechanism, to install extension; check Install Additional OSATE Components from Help menu. From there; you can install a more recent version of the plugin

closed 2020-06-28 by yoogx

nuaaysh, 2020-06-28:
Dear Prof. Jerome Hugues,
      Many thanks for your reply, now I get the latest version and the source doc of PolyORB-HI/Ada. But I test the example, it still reports an error:ADA.IO_EXCEPTIONS.USE_ERROR.(below)
      I try to google it or check the User's guide, But I can't find a solution.

      Could you please give me some suggestions?

Best wishes,

+========================== OCARINA BUG DETECTED =========================+
| Detected exception: ADA.IO_EXCEPTIONS.USE_ERROR                         |
| Error: copy of ".../ocarina_install/include/ocarina/runtime/polyorb-hi-ada/polyorb_hi-port_type_marshallers.ads" failed |
| Please refer to the User's Guide for more details.                      |
+=========================================================================+

raised ADA.IO_EXCEPTIONS.USE_ERROR : copy of ".../ocarina-build/ocarina_install/include/ocarina/runtime/polyorb-hi-ada/polyorb_hi-port_type_marshallers.ads" failed

Symbolic Traceback:

yoogx, 2020-06-28:
Which example are you testing? All PolyORB-HI/Ada examples shipped with the source code release are tested, see https://travis-ci.org/github/OpenAADL/ocarina

nuaaysh, 2020-06-28:
Hello, Prof. Jerome Hugues,

The example is from https://github.com/OpenAADL/polyorb-hi-ada/tree/master/examples/aadlv2/prime_numbers

yoogx, 2020-06-28:
Do not try randomly ... this example is meant to be compiled for Native platform,

You may want to change this from LEON_ORK to Native
'''
 processor the_processor
 properties
-  Deployment::Execution_Platform => LEON_ORK;
+  Deployment::Execution_Platform => Native;
 end the_processor;
'''

then you need to update Ocarina to have patches for this issue openaadl/ocarina#262
note they are independent of the bug you mentioned.

nuaaysh, 2020-06-28:
Many Thanks! I have modified the AADL model.

But now the error still exists: "Detected exception: ADA.IO_EXCEPTIONS.USE_ERROR", so I can't use the ocarina-osate plugin on my Ubuntu.  

Could you please give me some suggestions?

yoogx, 2020-06-28:
You did not tell me whether the file exists or not; I would recommend you try with GNAT Community Edition 2019 if you use GNAT from Ubuntu

nuaaysh, 2020-06-28:
Sorry, my mistake!
[image]
So please check the picture: The .abs exists and the ocarina terminal says there is an error: can't copy the abs.file! But I also get the target Ada code including the .abs file!!!

yoogx, 2020-07-10:
Issue #264 addressed a bug with GNAT CE 2020. This should now be corrected