OpenAADL/ocarina #286: Possible error that causes failures in automated build/test environment

nvcyc, 2021-02-08:
I'm creating this issue and providing some information that I'm examining in an attempt to explain why the pull request #284  passed the tests and failed when merged and hopefully fix a possible error in the automated build/test environment.

By looking at the automated build/test [log](https://github.com/OpenAADL/ocarina/pull/284/checks?check_run_id=1818152101) for #284, we can see that, at a high level, it has the following key stages that are related to this discussion:
1. Check out repository
2. Set up ocarina-builder 
3. Build

From the log, we can see that it does pull and check out the correct code with `/usr/bin/git checkout --progress --force refs/remotes/pull/284/merge` and yield the following message:
```
HEAD is now at a768dfa Merge 402cf43f62edb5340ac83ed330ec8f7905412ea8 into 5582e59fea10a212732c89c3a4e95a1879cd8bd4
```
which indicates that the current HEAD is `a768dfa`.

However, in the `Build` stage, the log shows that it runs `cd ocarina-build ; ./build_ocarina.sh --scenario=github` which pulls the latest code directly from the main branch on GitHub.
Provided there is no hidden arguments given in the above command, this runs the test against the latest code on GitHub (which is `5582e59`) rather than the `a768dfa` code from the pull request.

This speculation might be supported by the part of the log where it builds the code for archives.
In the same log for #284, it generates the following message when making the archive which attaches the master branch's hash value `5582e59` (possibly what the script gets from the currently checked-out branch) to the file name:

```
[2021-02-02-22:08] Configure for dist (Ocarina)    [PASSED]                                                       
tput: No value for $TERM and no -T specified
[2021-02-02-22:08] make dist (Ocarina)    [PASSED]                                              
  => Archive ocarina-v2017.1-496-g5582e59.tar.gz built in directory /home/runner/work/ocarina/ocarina/ocarina-build/ocarina
  => Source archive ready: ocarina-v2017.1-496-g5582e59-suite-src-20210202.tar.gz
rm: cannot remove 'ocarina-v2017.1-496-g5582e59-suite-src-20210202': No such file or directory
tput: No value for $TERM and no -T specified
[2021-02-02-22:08] extracting archive ocarina-v2017.1-496-g5582e59-suite-src-20210202.tar.gz    [PASSED]                                                                                                    
tput: No value for $TERM and no -T specified
[2021-02-02-22:08] DIST: configure (Ocarina)    [PASSED]                                                    
tput: No value for $TERM and no -T specified
[2021-02-02-22:08] DIST: make (Ocarina)    [PASSED]                                               
tput: No value for $TERM and no -T specified
[2021-02-02-22:09] DIST: make install-strip (Ocarina)    [PASSED]                                                             
tput: No value for $TERM and no -T specified
[2021-02-02-22:09] DIST: make distclean (Ocarina)    [PASSED]                                                         
tput: No value for $TERM and no -T specified
[2021-02-02-22:10] DIST: remove old archives    [PASSED]                                                    
Preparing archive: format .tgz
  => Archive ocarina-v2017.1-496-g5582e59-suite-linux-x86_64-20210202.tgz built in directory /home/runner/work/ocarina/ocarina/ocarina-build/ocarina
```

In short, it might be worth checking if the build stage's `cd ocarina-build ; ./build_ocarina.sh --scenario=github` needs to be corrected to use the repository/code that is already checked out from previous stage.

nvcyc, 2021-02-08:
A note for the pull request #284: the tests should fail for #284 as it contains a bug that is fixed in #285. This is true when #284 was merged and tested with failures.
If the above speculation is correct, the tests conducted in the pull request #284 were against the master branch which passed as expected.

yoogx, 2021-02-08:
Thanks for the analysis. This is most certainly correct. build_ocarina.sh can already fetch another repo, I will adjust the automated build accordingly.

yoogx, 2021-02-09:
This issue should now be solved, I had to tweak the scripts to fetch the right repository.

closed 2021-02-09 by yoogx