OpenAADL/ocarina #288: Support for the XML format from Cheddar 3.2 by the cheddar backend

nvcyc, 2021-02-22:
#### OCARINA VERSION:
```
root@05f8f72842cc:~/opt/ocarina-build# ocarina --version
Ocarina v2017.1-501-g402cf43 (Working Copy from r402cf43f)
Copyright (c) 2003-2009 Telecom ParisTech, 2010-2021 ESA & ISAE
Build date: Feb 05 2021 00:06:31
```

#### DESCRIPTION:
The `cheddar` backend generates a project file in XML that is intended for Cheddar to create a corresponding system environment for conducting further scheduling analysis (e.g., schedulability analysis).
However, the latest Cheddar version (`Cheddar 3.2`) doesn't seem to accept such an XML output from the `cheddar` backend in Ocarina. This XML output by `cheddar` backend is also different from the one that can be obtained from the Cheddar plug-in in Osate2.
Therefore, it is needed to clarify the following questions:
1. Which Cheddar version accepts the XML output generated by the current `cheddar` backend?
2. What is the plan for `cheddar` backend? Is the goal to support the latest Cheddar version (`Cheddar 3.2`)?

#### SAMPLE FIX/WORKAROUND:
I think supporting the latest Cheddar will be beneficial, so it is worth revising the existing `cheddar` backend or creating another backend version to support `Cheddar 3.2`, depending on your plan and preference.
Either way, the first step to supporting the latest Cheddar is to identify the right XML format.
Apparently the [cheddar.dtd](https://github.com/OpenAADL/ocarina/blob/master/resources/runtime/cheddar/cheddar.dtd) file in Ocarina's source tree is obsolete. 
Do you know where we can obtain the correct format?
I found a [cheddar_adl.dtd](http://beru.univ-brest.fr/svn/CHEDDAR/trunk/src/platypus/generated-code/ada/cheddar_adl.dtd) file in Cheddar's source tree but not sure if this is the correct one we can follow.

yoogx, 2021-02-24:
The current Cheddar backend supports Cheddar 2.x, 
Upgrading to Cheddar 3.2 is not that complicated. However Cheddar's author created a plugin for OSATE that exports to 3.2. I recommend using this one instead.

nvcyc, 2021-04-16:
I realized that the output project file generated by `cheddar` backend is compatible with Cheddar's current `cheddarkernel` binary that accepts `2.x` format, so it would be a good idea to keep this backend and have a separate backend for `Cheddar 3.2` if needed in the future.

Apart from that, I think it will be a good idea to add support of specifying custom output file name to the current `cheddar` backend like what we added for `aadl_xml` which enables testing and increases usability. I will have a patch for this purpose if you agree.

yoogx, 2021-04-16:
I am not sure what will be the future of cheddarkernel, but I am fine not touching the code base

Yes, you can propose a PR to specify a custom output file