OpenAADL/ocarina #291: In Osate 2.8, I have configured the path of Ocarina bin ，Why is it still showing Path to Ocarina is not setup

Lqs66, 2021-03-22:
In Osate 2.8, I have configured the path of Ocarina bin ，Why is it still showing Path to Ocarina is not setup

jjhugues, 2021-03-22:
I just confirmed it works in my setup. You should confirm you have proper permissions to the binary.

Lqs66, 2021-03-23:
> I just confirmed it works in my setup. You should confirm you have proper permissions to the binary.

Sorry, my computer knowledge is very poor, can you go into more detail?

jjhugues, 2021-03-23:
Depending on the OS you are using, check you have read/write permissions.
Also, clarify what you mean by "Path to Ocarina is not setup", is it in the preference pane or when running one of the Ocarina command?

Lqs66, 2021-03-23:
> when running one of the Ocarina command

thank you for your reply ，when running one of the Ocarina command，"Path to Ocarina is not setup" was shown

Lqs66, 2021-03-23:
> Depending on the OS you are using, check you have read/write permissions.
> Also, clarify what you mean by "Path to Ocarina is not setup", is it in the preference pane or when running one of the Ocarina command?
I have read/write permissions，but it doesn't work

jjhugues, 2021-03-23:
Can you make a screenshot of the preference panel that shows the path? enlarge the window to show the full path

Lqs66, 2021-03-23:
> Can you make a screenshot of the preference panel that shows the path? enlarge the window to show the full path

[image]

Lqs66, 2021-03-26:
> Can you make a screenshot of the preference panel that shows the path? enlarge the window to show the full path

I have uploaded the picture, can you see it? Can you give me some suggestions?

jjhugues, 2021-03-26:
As I said, it should work. Did you try running Ocarina from the command line? You used a very old version of Ocarina, I cannot guarantee it will run on your system

Lqs66, 2021-03-27:
> You used the darwin version, which is meant for macOS
> but you PATH looks like a Windows one, so you should at least install the windows version

I found that this is indeed the case, where should I find the windows version ？

yoogx, 2021-03-29:
There is no 2017.1 release version for Windows. I suggest you test on Linux and build it from source, this is fully documented in the ocarina-build repository.

Build for Windows are also supported, but not testeD.

closed 2021-03-29 by yoogx