OpenAADL/ocarina #297: Flow source inconsistency error occurs when extended component's declaration happens first

nvcyc, 2021-07-08:
#### OCARINA VERSION:
```
root@1c19889fbb07:~/opt/ocarina-build# ocarina --version
Ocarina v2017.1-520-g8af588c (Working Copy from r8af588c4)
Copyright (c) 2003-2009 Telecom ParisTech, 2010-2019 ESA & ISAE , 2019-2021 OpenAADL
Build date: Jun 28 2021 21:27:26
```

#### HOST MACHINE and OPERATING SYSTEM:
```
root@1c19889fbb07:~/opt/ocarina-build# uname -a
Linux 1c19889fbb07 5.4.0-60-generic #67~18.04.1-Ubuntu SMP Tue Jan 5 22:01:05 UTC 2021 x86_64 x86_64 x86_64 GNU/Linux
```

#### DESCRIPTION:

When the declaration of a component (a process component in my case), say `p2`, that extends a component `p1` is placed before the declaration of `p1`, the Ocarina's analyzer throws an error stating that `p1`'s flow sources in its flow implementation and its flow spec are inconsistent.

This error is not thrown if the order is inverse (i.e., `p1` is declared first, followed by `p2`.)

Note that this order of declaration issue is true when `p1` and `p2` are in the same AADL file or when they are in different files (the issue becomes the order of the AADL files given in the command line arguments.)

Generally, the latter condition (i.e., components in separate files) is what we are really concerned with as we don't typically assume (or impractical to assume) any order when listing the AADL files in the command line arguments.

#### REPEAT BY:

As a simple example, let's consider a system `deployment` with a process `p1` and a thread `t1` enclosed.
Assuming the AADL code below is stored in the file `test_order.aadl`, the aforementioned issue appears when running `ocarina -aadlv2 test_order.aadl`

```
package test_order
public

    data d1 end d1;

    system deployment
    end deployment;

    system implementation deployment.impl
        subcomponents
            p1: process p1.impl;
    end deployment.impl;

    -- The extended component
    -- The error is not thrown if these declarations are placed after p1.
    process p2
    extends p1
    end p2;

    process implementation p2.impl
    extends p1.impl
    end p2.impl;
    -- End of extend component

    process p1
        features
            out_port: out data port d1;
        flows
            flow_out: flow source out_port;
    end p1;

    process implementation p1.impl
        subcomponents
            t1: thread t1.impl;
        connections
            conn_out: port t1.out_port -> out_port;
        flows
            flow_out: flow source t1.flow_out -> conn_out -> out_port;   -- where Ocarina throws an error against
    end p1.impl;

    thread t1
        features
            out_port: out data port d1;
        flows
            flow_out: flow source out_port;
    end t1;

    thread implementation t1.impl
    end t1.impl;

end test_order;
```

which throws the error
```
test_order.aadl:41:13: flow_out (flow implementation)  and its corresponding flow spec have different sources
Cannot analyze AADL specifications
```

The error does not occur when removing the declaration of `p2` or moving it to after `p1.impl`'s declaration.

yoogx, 2021-07-09:
Thanks. This is indeed a funny bug and quite unexpected since the order of declaration should not matter here.

yoogx, 2021-07-18:
Thanks for the patch, greatly appreciated

closed 2021-07-18 by yoogx

nvcyc, 2021-07-19:
Just to confirm with you, the PL #298 seems to be closed without being merged.
Are you planning to add the commit later manually or is there anything else I can help with this issue?
Thanks!

yoogx, 2021-07-19:
My bad, error in the GUI