OpenAADL/ocarina #303: if i want to run it on centos8, which version of automake should i install ?

Liufengyuann, 2021-09-08:
OCARINA VERSION:
  ocarina version = v2021.1 

 HOST MACHINE and OPERATING SYSTEM:
Linux hadoop01 4.18.0-305.12.1.el8_4.x86_64 #1 SMP Wed Aug 11 01:59:55 UTC 2021 x86_64 x86_64 x86_64 GNU/Linux

    DESCRIPTION:
When i run the instruction ”./configure“ 
But it had the error like is : 
`./configure: line 1884: syntax error near unexpected token `1.9'
./configure: line 1884: `AM_INIT_AUTOMAKE(1.9 tar-pax)'
`
It likes that the version of automake is wrong, so i want to know which of the autoconf`s version i should install? maybe 1.9? but i cant find the version in the internet...

What`s more i want to know which version of gnat is compatible with the ocarina ?
I`m not  good at this ,  hope get your answer ， thank you very much !

yoogx, 2021-09-08:
This is what I use

➜  polyorb-hi-c git:(master) ✗ automake --version
automake (GNU automake) 1.16.3

jjhugues, 2021-10-04:
No feedback, closing

closed 2021-10-04 by jjhugues