OpenAADL/ocarina #304: Backends: fatal error : This OUT port is not connected to any destination？

Liufengyuann, 2021-09-27:
Hello, when I run the **ping** example from polyorb_hi_c ,but i get the fatal error like this :

[image]

but the out port of thread P will be a subcomponent of process A, and thread Q will be a subcomponent of process B.
[image]

process A will connected to process B ,so the OUT PORT of Thread P will connected to Thread Q, but this error occurred.

So how can I solve this error? Hope I can get your help， thanks!

yoogx, 2021-09-27:
I would need more details on the version of Ocarina you are using; and which ping example you are using.
All examples are tested nightly and are known to work properly

Liufengyuann, 2021-09-27:
Ocarina version I`m using:
[image]
and the example path is 
`ocarina/ocarina-build/ocarina_install/examples/ocarina/polyorb-hi-c/aadlv2/ping
`
Can you make some suggestions to solve the problem？ thank you so much!

Liufengyuann, 2021-09-27:
Ocarina version Im using: 
 
[image]

and the example path is ocarina/ocarina-build/ocarina_install/examples/ocarina/polyorb-hi-c/aadlv2/ping
`
Can you make some suggestions to solve the problem？ thank you so much!

yoogx, 2021-09-27:
AFAICT it works.

Go in `ocarina/ocarina-build/ocarina_install/examples/ocarina/polyorb-hi-c/aadlv2/ping`and type `make build-all`. This is the test I run during regression tests. It compiles this example in various configurations. I do not test through OSATE.

In this example, the two processes are connected. But you changed it as the line numbers do not match. Also, the code does not match. I do not use "conn" as a name for the connections in this file. OSATE may have reported an error here, I have to port this example to OSATE, Ocarina is a bit lax here.

So first test the original example from the command line

Liufengyuann, 2021-09-28:
But the dir does not have a makefile, so maybe i can not run `make build-all` 
[image]
And when i type `ocarina -p ping.aadl` to parse the ping.aadl ,I  got these errors
[image]
Maybe the instruction i used is not correct ? Ping.aadl in the example like following:
 `--  This model completes the PING example by adding deployment
--  information.
--
--  In this deployment, two nodes are defined, one task per node.

package PING
public

  with Software;
  with Deployment;
  with ocarina_drivers;
  with ocarina_buses;

  ---------------
  -- Processor --
  ---------------

  processor the_processor
  properties
    Deployment::location           => "127.0.0.1";
    Deployment::Execution_Platform => Native;
    Priority_Range                 => 0 .. 255;
    Scheduling_Protocol => (POSIX_1003_Highest_Priority_First_Protocol);
  end the_processor;

  ---------
  -- Bus --
  ---------

  bus Ethernet_Bus
  properties
    Deployment::Transport_API => BSD_Sockets;
  end Ethernet_Bus;

  ---------------
  -- Processes --
  ---------------

  process A
  features
    Out_Port : out event data port Software::Simple_Type;
  end A;

  process implementation A.Impl
  subcomponents
    Pinger        : thread Software::P.Impl;
  connections
    port Pinger.Data_Source -> Out_Port;
  end A.Impl;

  process B
  features
    In_Port  : in event data port Software::Simple_Type;
  end B;

  process implementation B.Impl
  subcomponents
    Ping_Me        : thread Software::Q.Impl;
  connections
    port In_Port -> Ping_Me.Data_Sink;
  end B.Impl;

  process singleProcess
  end singleProcess;

  process implementation singleProcess.Impl
  subcomponents
    Pinger        : thread Software::P.Impl;
    Ping_Me       : thread  Software::Q.Impl;
  connections
    port Pinger.Data_Source -> Ping_Me.Data_Sink;
  end singleProcess.Impl;

  ------------
  -- System --
  ------------

  system PING
  end PING;

  system implementation PING.Impl
  subcomponents
    Node_A : process A.Impl;
    Node_B : process B.Impl;

    Device_A : device ocarina_drivers::generic_sockets_ip.pohic
    {Source_Text => ("devicesconf.c");};
    Device_B : device ocarina_drivers::generic_sockets_ip.pohic
    {Source_Text => ("devicesconf.c");};

    CPU_A : processor the_processor;
    CPU_B : processor the_processor;
    the_bus : bus ocarina_buses::ip.i;
  connections
    bus access the_bus -> Device_A.link;
    bus access the_bus -> Device_B.link;

    port Node_A.Out_Port -> Node_B.In_Port
    {Actual_Connection_Binding => (reference (the_bus));};
  properties
    actual_processor_binding => reference (CPU_A) applies to Node_A;
    actual_processor_binding => reference (CPU_B) applies to Node_B;
    actual_processor_binding => (reference (CPU_A)) applies to Device_A;
    actual_processor_binding => (reference (CPU_B)) applies to Device_B;
  end PING.Impl;

  system implementation PING.Local
  subcomponents
    ping : process singleProcess.Impl;
    CPU : processor the_processor;

  properties
    actual_processor_binding => (reference (CPU)) applies to ping;
  end PING.Local;

  system implementation PING.Xenomai
     extends PING.Local
  properties
    Deployment::Execution_Platform => linux32_xenomai_native applies to CPU;
  end PING.Xenomai;

  system implementation PING.RTEMS
     extends PING.Local
  properties
     Deployment::Execution_platform => LEON_RTEMS applies to CPU ;
  end PING.RTEMS;

end PING;`

yoogx, 2021-09-28:
You may want to read the doc first, `ocarina -p` parses one file, but you need extra parameters to indicate which version of AADL you want to use, etc.

PolyORB-HI/C examples must be ported to OSATE, a few elements are missing like names for connections. OSATE is stricter than it used to be on this. Only the examples from AADLib have been ported to OSATE

You are right there is no makefile, it seems you used `build_ocarina.sh`to compil, and I missed that you were referencing the install directory and not the build one. If so go, in the source directory; (path from my install, you may adapt this° `ocarina-build/ocarina/resources/runtime/polyorb-hi-c/examples/aadlv2/ping`and run `make build-all`

alternatively; from the command line: `ocarina -bx scenario.aadl` will generate code and compile it. It does not depend on any makefile and should work in your install directory

Liufengyuann, 2021-09-28:
Thank you!!! I have successfully run it!! Thank you so much~
My research topic is the conversion of aadl to C language. I would like to know in which part of your code the aadl parsing process is implemented? Thank you so much for your help!

jjhugues, 2021-09-28:
Glad it worked. Ocarina is a compiler, so review the code: `frontends` has the parsing, `backends` the code generation part for the multiple targets we support. Code generation simply instantiate the PolyORB-HI/C middleware

closed 2021-09-28 by jjhugues