OpenAADL/ocarina #307: How can I view the structure of xml_tree generated by aadl?

Liufengyuann, 2021-10-14:
NOTE: Only file GitHub issues for bugs and feature requests. All other topics will be closed.

For bugs or installation issues, please provide the following information. The more information you provide, the more easily we will be able to offer help and advice.

Also, make your Subject: line as descriptive as possible.  Subjects like "Ocarina bug" or "bug report" are not helpful!

Please, also clean this template to leave only relevant bits.

8<----------8<----------8<----------8<----------8<----------8<----------8<----

    OCARINA VERSION:
[Please provide the complete output of "ocarina --version".]

    HOST MACHINE and OPERATING SYSTEM:

[Provide the output of "uname -a"]

    COMPILER VERSION

[Please provide the complete version string from "gnatls -v" if you try to compile Ocarina or PolyORB-HI/Ada, or "gcc --version" for PolyORB-HI/C.]

    DESCRIPTION:
I want to know the xml file generated from aadl file, how can i find it? thanks for your help~!

    REPEAT BY:

[What you did to get the error; include test program or session transcript if at all possible; detail your application topology if it uses several nodes, and describe the expected results; include the configuration file you use, after stripping all comments; include all information regarding any other middleware you use. ]

    SAMPLE FIX/WORKAROUND:

[If available]

jjhugues, 2021-10-14:
You need to use the aadl_xml backend. From the command line, just use `-g aadl_xml` 

If you look at https://github.com/openaadl/aadlib, you can for instance test in `examples/rma` this backend. Just type `make aadl_xml`

Liufengyuann, 2021-10-14:
I get it! thanks~!!