OpenAADL/ocarina #313: whether can ignore 'source_text' in subprogram?

ibertli, 2023-04-10:
Hello Sir, i have a question asking for your help. Based on my understanding，we cannot ignore 'source_text' in subprogram, but i found in '/ocarina-build/ocarina/doc/source/ping/devices.aadl' at line21 where the special subprogram has no 'source_text'. It confused me.
thanks for your replay!

yoogx, 2023-04-10:
What do you mean by "we cannot ignore source_text in source program"? A model is perfectly valid if subprograms do not define source_text. However it may not be correct for a specific model processing if absent. For instance, scheduling analysis does not need this property defined, but code generation does. Since code generation may not consider device, it is fine.

ibertli, 2023-04-11:
> What do you mean by "we cannot ignore source_text in source program"? A model is perfectly valid if subprograms do not define source_text. However it may not be correct for a specific model [...]

I do not understand. If i do code generation work use software component without 'source_text' in subprogram, how can i confirm which file to execute when call a subprogram and where is the function 'subprogram_b'? Example like this:
****************
SUBPROGRAM subprogram_b
features
  subb_port: in parameter Base_Types::integer;
PROPERTIES
  SOURCE_LANGUAGE => (C);
  SOURCE_NAME => "subprogram_b";
END subprogram_b;
****************

yoogx, 2023-04-11:
I suggest you study the provided examples. 

As I said; 
- your example is perfectly valid if you perform scheduling. The Source_Text property is not necessar.
- your example is invalid for code generation: the name of the C file is missing and you cannot complete code generation

ibertli, 2023-04-12:
I know what's your mean. Aimed at code geration, i cannot ignore 'source_text'。Thanks very much!