OpenAADL/ocarina #314: bug: can not generate data type with 'string'

ibertli, 2023-04-13:
hello sir, i find a bug, when i define a data type using 'string' representation, ocarina raise error "Impossible to get the data size of this data", then i check the source code and find that when Data_Representation equals to Data_String, Value_Node was not updated in '../ocarina/src/backends/ast_c/ocarina-backends-c_tree-nutils.adb'. May be you can confirm whether it is a bug. 
********
--  The example like this:
data person_name
properties
  Data_Model::Data_Representation => string;
  Data_Model::Dimension => (10);
end person_name;

subprogram print_a
features
  port2: in parameter person_name;
end print_a;
********

and the source code like this:
*******
         when Data_String | Data_Wide_String =>
            Value_UUL := Dimension (1);
*******
Thanks for your reply!

yoogx, 2023-04-14:
Thanks for reporting. Yes, this support has never been implemented. 
A work-around is to use an array of char.

ibertli, 2023-04-17:
> Thanks for reporting. Yes, this support has never been implemented. A work-around is to use an array of char.
> 
> I will look into implementing this feature when time permits.

got it, wait for the update. Thanks!