OpenAADL/ocarina #315: how to import custom Property Set?

ibertli, 2023-04-18:
hello sir, i got a problem want to ask for your help. When i import a custom property set use 'with' statement, and run ocarina, it throw an error "can not open file xx.aadl"，i have try some different methods, for example: move 'my_pro.aadl' to the path same to 'Data_Model.aadl', or use '-I ARG', but all of that are useless. i want to know how to correctly import custom property set. 
demo like this:
**************
--  my_pro.aadl
property set bst_pro is
  CoreNum: aadlinteger applies to (thread);
end bst_pro;

--  test1.aadl
PACKAGE rmaaadl
PUBLIC
WITH deployment;
with Data_Model;
with Base_Types;
with my_pro;
...
end rmaaadl
**************
Thanks for your reply!

yoogx, 2023-04-18:
How do you use Ocarina? which command line argument do you pass?

ibertli, 2023-04-18:
> 
I use ocarina in Ubuntu 20.04, and test1.aadl only has one system named 'rma.c', i use the command to generate C code like this:
"./ocarina -aadlv2 /home/ubuntu/Documents/ocarina-build/ocarina_install/share/ocarina/AADLv2/ -f -disable-annexes=all -r rmaaadl::rma.c -g polyorb_hi_c -o /home/ubuntu/Documents/aadl_demo/ /home/ubuntu/Documents/aadl_demo/test.aadl"

If i do not import 'my_pro' use "with my_pro" statement in 'test1.aadl', ocarina can generate C code successfully, but if i try to  import 'my_pro' and run ocarina with command like this:
********
"./ocarina -aadlv2 /home/ubuntu/Documents/ocarina-build/ocarina_install/share/ocarina/AADLv2/ -f **-y** -disable-annexes=all -r rmaaadl::rma.c -g polyorb_hi_c -o /home/ubuntu/Documents/aadl_demo/ /home/ubuntu/Documents/aadl_demo/test.aadl"
********
it will throw an error "undefined cannot find file my_pro.aadl". It seems that ocarina not find the file in a special path.

yoogx, 2023-04-19:
From --help
 -I ARG               Add ARG to the directory search list

I guess you should have done this

"./ocarina -aadlv2 **-I** /home/ubuntu/Documents/ocarina-build/ocarina_install/share/ocarina/AADLv2/ -f -y -disable-annexes=all -r rmaaadl::rma.c -g polyorb_hi_c -o /home/ubuntu/Documents/aadl_demo/ /home/ubuntu/Documents/aadl_demo/test.aadl"

ibertli, 2023-04-19:
> 

Actually, i have tried like this, but still got an error "cannot find file my_pro.aadl"。Do you use the command line above without reporting any errors？

sei-jhugues, 2023-04-19:
Yes

➜  issue_315 ocarina -v -y -aadlv2 test1.aadl 
Ocarina v2017.1-510-gf2c500f (Working Copy from rf2c500fe)
Copyright (c) 2003-2009 Telecom ParisTech, 2010-2019 ESA & ISAE , 2019-2021 OpenAADL
Build date: Feb 26 2021 13:15:24
Loading file test1.aadl
Loading file my_pro.aadl
Loading of all files done
Model parsing: completed

➜  issue_315 cat *.aadl                         
property set my_pro is
CoreNum: aadlinteger applies to (thread);
end my_pro;
PACKAGE test1
PUBLIC
with my_pro;

end test1;

sei-jhugues, 2023-04-19:
> > 
> 
> Actually, i have tried like this, but still got an error "cannot find file my_pro.aadl"。Do you use the command line above without reporting any errors？

In the command above, add -v to see what Ocarina does. Note that the search directories are 1/ the directory from which you invoke Ocarina, 2/ the directory of default property sets, 3/ whatever is specified in -I flags.

ibertli, 2023-04-19:
> In the command above, add -v to see what Ocarina does. Note that the search directories are 1/ the directory from which you invoke Ocarina, 2/ the directory of default property sets, 3/ whatever is [...]

Thanks very much, i have resolved the problem base on your suggestion!

closed 2023-04-20 by yoogx