OpenAADL/ocarina #316: how to import file whilefile_name is not same to the package_name

ibertli, 2023-05-18:
hello sir:
  I got a question and need your help.
  If i import a file while file_name is not same to the package_name with ocarina. For example:
  ************ 
  -- file: pkg_2.aadl
  package pkg_1
    -- write something here
  end pkg_1;
  ************

Bsed on aadl specification, i should use syntax rule "import_declaration ::= with (package_name | property_set_identifier)" to import the aboved-mentioned file. But when i import the file in anthoer aadl file and run ocarina use "- I", it throw an error "cannot not find pkg_1.aadl", it makes me confused. Whether i have to define a file and file_name samed to the package_name ot property_set_name?

************
-- file test.aadl
package test
public
with pkg_1;  -- this will throw an error
with pkg_2;  -- this does not match the aadl specification
end test;
************

jjhugues, 2023-05-18:
You can either run ```ocarina -aadlv2 test.aadl pkg_2.aadl```. This will load all files and parse. You will get the following error message

```
test.aadl:5:06: pkg_2 (identifier) is not a package or a property set visible or existing
Cannot analyze AADL specifications
```

after changing ```pkg_2.aadl``` to

```
-- file: pkg_2.aadl
package pkg_1
public
  system s end s;
-- write something here
end pkg_1; 
```

or you can run ```ocarina -aadlv2 -y test.aadl```. ```-y``` will load files based on the package name. Then you get
```Cannot find file pkg_1.aadl``` as expected

OSATE and Ocarina enforces the rules that the file name must match the package name to support lazy loading. (option 2 here).  ```A::b``` is translated into ```a-b.aadl```

ibertli, 2023-05-19:
> You can either run `ocarina -aadlv2 test.aadl pkg_2.aadl`. This will load all files and parse. You will get the following error message
>
> ```
> test.aadl:5:06: pkg_2 (identifier) is not a package or a [...]

Hahhhh, i know what's your mean. Thanks very much!